bromagosa/Snap4Arduino #1: Cloud signup doesn't work

edutec, 2016-02-02:
_From @maprzybylla on January 21, 2016 10:1_

When I try to sign up I get a message: 
"No such file - SnapCloudLocalSignUp.
An e-mail with your password has been sent to the address provided."
[image]

_Copied from original issue: edutec/Snap4Arduino-old-huge#74_

bromagosa, 2016-02-15:
@maprzybylla is this still an issue? It does work for me.

maprzybylla, 2016-02-15:
Yes, still is an issue in 1.0.9 beta (Mac OS) - just tried it and got the same message, but no e-mail. In addition, Snap4Arduino crasehd - heres the report:

Process:               nwjs [1365]
Path:                  /Users/USER/Downloads/Snap4Arduino.app/Contents/MacOS/nwjs
Identifier:            io.nwjs.nw
Version:               48.0.2564.97 (2564.97)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           nwjs [1365]
User ID:               501

Date/Time:             2016-02-15 12:15:37.133 +0100
OS Version:            Mac OS X 10.11.3 (15D21)
Report Version:        11
Anonymous UUID:        D0AE113A-F32F-235B-5CA7-13A045643019

Time Awake Since Boot: 10000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Item to be removed is not in the menu in the first place'
terminating with uncaught exception of type NSException
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff9649eae2 __exceptionPreprocess + 178
1   nwjs Framework                      0x000000010b7df56e ChromeMain + 580158
2   libobjc.A.dylib                     0x00007fff8235af7e objc_exception_throw + 48
3   CoreFoundation                      0x00007fff9649e8ba +[NSException raise:format:arguments:] + 106
4   Foundation                          0x00007fff80b5c88c -[NSAssertionHandler handleFailureInMethod:object:file:lineNumber:description:] + 198
5   AppKit                              0x00007fff91f52c7f -[NSMenu removeItem:] + 300
6   nwjs Framework                      0x000000010e657c45 _ZN2v88UnlockerD1Ev + 19941381
7   nwjs Framework                      0x000000010e65765f _ZN2v88UnlockerD1Ev + 19939871
8   nwjs Framework                      0x000000010e656fd9 _ZN2v88UnlockerD1Ev + 19938201
9   CoreFoundation                      0x00007fff9645870c __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
10  CoreFoundation                      0x00007fff9645867f ___CFXRegistrationPost_block_invoke + 63
11  CoreFoundation                      0x00007fff96457d47 _CFXRegistrationPost + 407
12  CoreFoundation                      0x00007fff96457ab2 ___CFXNotificationPost_block_invoke + 50
13  CoreFoundation                      0x00007fff96451d42 -[_CFXNotificationRegistrar find:object:observer:enumerator:] + 1922
14  CoreFoundation                      0x00007fff96340145 _CFXNotificationPost + 693
15  Foundation                          0x00007fff80aa1921 -[NSNotificationCenter postNotificationName:object:userInfo:] + 66
16  AppKit                              0x00007fff91ffb7d6 endKeyAndMain + 99
17  AppKit                              0x00007fff91e0efac -[NSApplication sendEvent:] + 6991
18  nwjs Framework                      0x000000010b7dff5d ChromeMain + 582701
19  nwjs Framework                      0x000000010bc6606a ChromeMain + 5326138
20  nwjs Framework                      0x000000010b7dfc18 ChromeMain + 581864
21  AppKit                              0x00007fff91d4103a -[NSApplication run] + 796
22  nwjs Framework                      0x000000010bc42941 ChromeMain + 5180945
23  nwjs Framework                      0x000000010bc42044 ChromeMain + 5178644
24  nwjs Framework                      0x000000010bc82273 ChromeMain + 5441347
25  nwjs Framework                      0x000000010b7e4b15 ChromeMain + 602085
26  nwjs Framework                      0x000000010e9ed057 _ZN2v88UnlockerD1Ev + 23698455
27  nwjs Framework                      0x000000010e9ef8d2 _ZN2v88UnlockerD1Ev + 23708818
28  nwjs Framework                      0x000000010e9e8afd _ZN2v88UnlockerD1Ev + 23680701
29  nwjs Framework                      0x000000010bbfd6b8 ChromeMain + 4897672
30  nwjs Framework                      0x000000010bbfcb56 ChromeMain + 4894758
31  nwjs Framework                      0x000000010b751b72 ChromeMain + 66
32  nwjs                                0x000000010b4ead52 main + 530
33  nwjs                                0x000000010b4eab34 nwjs + 2868
34  ???                                 0x0000000000000002 0x0 + 2

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff822c3002 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff94b4e5c5 pthread_kill + 90
[... 760 lines cut ...]
__UNICODE                          552K        2 
mapped file                       80.7M       51 
shared memory                     16.3M       11 
===========                     =======  ======= 
TOTAL                            894.4M      819 

Model: MacBookPro11,3, BootROM MBP112.0138.B16, 4 processors, Intel Core i7, 2,6 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x02FE, -
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x02FE, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.94.136.1a1)
Bluetooth: Version 4.4.3f4 16616, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en3
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM0512F, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: External USB 3.0
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Keyboard Hub
USB Device: Apple Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 3, 5.5

bromagosa, 2016-02-15:
Oh wait, I mixed things up! It's logging in that does work okay, I never tried signing up.

I'll take a look at this, thanks :)

fabianhemmert, 2016-04-18:
Same here, signing up does not work for me (OSX, 1.1.2-beta).

HyperDuino, 2016-04-23:
Same thing here.  doesn't work for me on OSX.  ("Hi!" Mareen)  :)

bromagosa, 2016-05-30:
The problem seemed to come from the fact that Snap! mods use a different cloud URL, and the service behind it doesn't seem to have a signup endpoint. Same problem was preventing "reset password" from working.

My temporary -or not- solution is to change the URL to the original Snap! one before signing up, then reverting the change afterwards.

commit abd5034f (2016-05-30): fixes #1
closed 2016-05-30 by bromagosa