bromagosa/Snap4Arduino #17: HTTP protocol

edutec, 2016-02-02:
_From @bromagosa on July 31, 2014 8:2_

Just like in http://s4a.cat/#android under HTTP, it'd be swell to have an HTTP protocol for remote control.

_Copied from original issue: edutec/Snap4Arduino-old-huge#13_

edutec, 2016-02-02:
_From @elaval on July 31, 2014 9:13_

Some months ago I explored building a RESTfull web API for "talking" to an Arduino through http:

https://github.com/elaval/webduino

It created an httpserver and received REST commands to get/set pins data, and was built on top of johnny-five.js & other node modules.

I could try to get back to it and explore an API for Snap4Arduino.

edutec, 2016-02-02:
_From @bromagosa on July 31, 2014 9:36_

The idea is to let the user control Snap, not only the Arduino functionalities. In S4A we achieved this by creating a protocol that roughly worked [like this](http://s4a.cat/#android):

- You could send a message from the client to Scratch, and then access it with a [When message [whatever] received do] block.
- You could choose to read a Scratch variable from the client. For this, your variables had to follow a convention:
  - +varName would be readable and writeable from the client
  - -varName would be readable from the client
  - varName would not be seen from the client
- Same goes for messages already existing in Scratch, this convention applies:
  - +messageName would be readable from the client
  - messageName would be private to Scratch

(We can of course think of a better protocol)

edutec, 2016-02-02:
_From @chrisclaynz on June 26, 2015 6:24_

I'm new to all this but currently trying to use VS1838b IR receiver to create a simple robot that I can control with a tv remote. Is this currently possible with snap4arduino? If so - how? If not - why and what might be the best workaround?

bromagosa, 2016-02-08:
Based on how Scratch used to work, I've implemented a tiny server, listening on port 420001, that follows the exact Scratch specifications:

- **http://your-host:42001/send-vars** → Returns a list of all global variables and their values in the same format Scratch used to do it.
- **http://your-host:42001/send-messages** → Returns a list of all messages in the same format Scratch used to do it.
- **http://your-host:42001/broadcast=[message name]** → Broadcasts a message
- **http://your-host:42001/vars-update=[variable name]=[variable value]** → Sets an existing global variable to a new value

I've purposely ignored *sensor-update* because there is no generic *sensor* block in Snap!, so this one made no sense here... besides, messages and variables are way enough for all kinds of remote control and communications.

closed 2016-02-08 by bromagosa