bromagosa/Snap4Arduino #9: Communication problems with Arduino on MacOS when there are other bluetooth devices

edutec, 2016-02-02:
_From @maprzybylla on October 28, 2014 12:53_

With the beta release I'm no longer able to connect my Arduino Uno to Snap4Arduino. I tried it with different Unos, using both the OldStandardFirmata and the current one. I can communicate with the Board in alpha version 1.0.6. and in Windows and also tested with the firmata_test program provided by firmata.org
[image]
[image]

_Copied from original issue: edutec/Snap4Arduino-old-huge#39_

edutec, 2016-02-02:
_From @elaval on October 28, 2014 17:14_

Hi @maprzybylla,

I understand that you were never able to connect to the Uno on the Mac version 1.0 beta or was it that you could connect once and then not again?

edutec, 2016-02-02:
_From @elaval on October 28, 2014 17:25_

I have just tried with an Arduino UNO en my Mac and could connect.  Let's try to identify the conditions that make this error to happen.

If you quit the program, unplug the USB cable, plug the USB cable and open the program again ... Can you connect? 

[image]

edutec, 2016-02-02:
_From @maprzybylla on October 28, 2014 20:16_

No, I tried this but it doesn't help. I tried the same thing on two different Macs, both running OS X 10.10. Maybe that's the key? If I remember correctly, I never was able to connect in the beta build, only in the alpha versions.

edutec, 2016-02-02:
_From @elaval on October 28, 2014 20:59_

I am puzzled .... just tried on a OSX 10.10 and connected normally. There must be a pattern that explains it, but I don't have any hypothesis now.   I wish  I could reproduce the problem!!

edutec, 2016-02-02:
_From @maprzybylla on October 28, 2014 21:19_

Okay, now I found a Mac, also running OSX 10.10, where it works. I have no idea why. On the Mac where it works now, I have no other software installed that connects to arduino. On both of the Macs where it didn't work I've got various versions of the Arduino IDE, S4A, Snap4Arduino, and the like. Could that be a problem? However, I didn't start any of the other environments when trying to connect to the board in Snap4Arduino. On this Mac here where it works, I didn't install any drivers, while on the other Macs with the first installation of the Arduino environment possibly drivers were installed.

edutec, 2016-02-02:
_From @bromagosa on October 28, 2014 21:41_

It's a long shot, but it could be a problem related to having ran older versions of Snap4Arduino in the same computer. Sometimes node-webkit keeps stuff in cache, which may interfere with newer versions.

I'll look around to see how this cache can be cleared and get back to you.

edutec, 2016-02-02:
_From @elaval on October 29, 2014 11:54_

It could be something related to conflicting configurations stored in the Local Storage such as the problem we had on #26.  But I am not convinced ... the error screen shot posted by @maprzybylla indicates that the problem is on s4aObjects.js line 143 (Cannot read property 'sp' of undefined).

This is the code involved in the error:

```javascript
		myself.arduino.board = new world.Arduino.firmata.Board(port, function(err) { 
			if (!err) { 
				
				// Clear timeout to avoid problems if connection is closed before timeout is completed
				clearTimeout(myself.arduino.connectionTimeout); 

				myself.arduino.board.sp.on('disconnect', myself.arduino.disconnectHandler);
				myself.arduino.board.sp.on('close', myself.arduino.closeHandler);
				myself.arduino.board.sp.on('error', myself.arduino.errorHandler);
```

The error (line 143) occurs at 
```javascript
				myself.arduino.board.sp.on('disconnect', myself.arduino.disconnectHandler);
```

At that point: 
- a new Board object has been created and the callback function was called.
- the callback function is called with no error (a firmata communication was established through the serial port)
- BUT when we try to assign the 'disconnect' event handler to the serialport ('sp') we find that myself.arduino.board is undefined (and therefore myself.arduino.board.sp does not exist).

This does not make sense and I don't get why it could be happening.

Even though I am not convinced that it is related to the configuration in the local storage ... let's try to explore that path (and discard or find it).   Since I don't have a Mac where the connection problem happens I would like to ask for help from @maprzybylla (Are you willing to explore this issue further in your Mac?)

I have just left a "debug" version of Snap4Arduino 1.0 beta for Mac at this url (It will be available for 1 day):

https://elaval.s3.amazonaws.com/snap4arduino/Snap4Arduino-1Beta-Debug.zip?AWSAccessKeyId=AKIAJIKQLM7LQ2LXVNSA&Expires=1414669131&Signature=pvnflVacO%2FiAj1XZtX0niZjBSBo%3D

With this version we have access to debugging tools similar to the ones found in Chrome:
[image]

If you click in the DevTools cog (marked with a circle in the previous image) you will have access to a new window, where you can click on the "Resources" tab and expand the Local Storage  "file://" Key-Value settings.  
[image]
Could you post an image of the settings in the Mac that does not connect?

Thanks!

edutec, 2016-02-02:
_From @maprzybylla on October 29, 2014 12:23_

Hi, sure I'll help exploring the issue. This it the screenshot you asked for:
[image]

edutec, 2016-02-02:
_From @elaval on October 29, 2014 12:43_

Ok ... so you have stored 8 projects, the language setting ('en') and the zoom setting (1).  I do have the same settings for language & zoom and I can connect.  I don't think that the saved projects could be affecting the connection but we might want to discard this possibility.

You could explore to delete all settings and then close & reopen the program to see if this has any effect in the connection.  BUT (ATTENTION - ACHTUNG) this implies that you would loose all those 8 projects that you have saved (you won't see them even if you open the Alpha version).  If you can spare them (or export the projects as XML files so that you can have a backup) it would be good to see how it works  with a "clean" configuration.

To delete them, click on each the Key/Value and then click on the X.

edutec, 2016-02-02:
_From @elaval on October 29, 2014 13:10_

Well ... and since we are already in the "backstage" of the program we could also try some debugging putting a breakpoint in the line that generates the error.
[image]
In the "Sources" tab, look of s4aObjects.js and click on the number for line 143 to set the breakpoint.

If you now try to connect your Arduino, the program should stop at that line and we can explore the content of the object  "myself.arduino.board.sp"

myself: should be a reference to the sprite we are working with
arduino: is an object we assign to the sprite to store all arduino related properties
board: is an object that corresponds to the connected board and should have been created (but the error happens because it is undefined)
sp: is an object (part of board) for handling serialport communication

(Board object is part of the firmata.js open source library, and sp - Serialport - is part of serialport.js open source library)

This is what I see when I look at the variables:
[image]
[image]

edutec, 2016-02-02:
_From @maprzybylla on October 29, 2014 13:17_

No success. I deleted all projects and then in a second attempt also language and zoom settings. 
When I look into the variables, board is undefined:
[image]
[image]

edutec, 2016-02-02:
_From @maprzybylla on October 29, 2014 13:32_

I'm not sure if this is related, but I also get an uncaught error when I click the connect button and no board is physically connected with a cable (again this happens only on those Macs where it does not connect). This is the error message: 
"Uncaught node.js Error
Error: Cannot open /dev/cu.HC-06-DevB
at Error (native)"

edutec, 2016-02-02:
> No success. I deleted all projects and then in a second attempt also language and boom settings.

_From @elaval on October 29, 2014 13:36_

Later today I will try using an event instead of the callback function and see if we can avoid having the board object undefined.

edutec, 2016-02-02:
_From @bromagosa on October 29, 2014 14:6_

hmm... DevB sounds like it's trying to connect to some random bluetooth
device... do you have any bluetooth devices paired with the computer that
gives this error?

edutec, 2016-02-02:
_From @elaval on October 29, 2014 14:8_

@maprzybylla YES .. the "Cannot open /dev/cu.HC-06-DevB" is an important lead!!

It is trying to connect to the seraiport "/dev/cu.HC-06-DevB" and in my machines it is always with the format "/dev/cu.usbmodemfd121".

Originally we were checking only serial ports that contained /usb|acm|^com/i as candidates to have and arduino connected by USB cable, but a few commits ago we also included /usb|DevB|rfcomm|acm|^com/i to include also Bluetooth (@bromagosa do you remember this?).

So I guess that we are wrongly trying to connect to the port /dev/cu.HC-06-DevB and and not to the one that actually has the Arduino.

Are the Macs that are not working a different type than the one that dies work (i.e. Desktop v/s laptop).  I bet that the ones that do not work have some additional device (any Bluetooth?).
 
Please try the following in the Debug window:
- Open the console tab
- Execute "world.Arduino.serialport.list(function(err, list) {console.log(list)})"

You will get a list of the serial ports in the machine.  Try to execute the command with the arduino connected and with the arduino disconnected and tell me which are the ports in the list in each case.

Mine are:
(With arduino)
[image]
(Without arduino)
[image]

edutec, 2016-02-02:
_From @maprzybylla on October 29, 2014 14:27_

The two computers that don't connect are an iMac and a MacBook Pro, the one that does connect is an older MacBookPro (of which I don't have the details in mind). All of today's test were executed on this MacBook Pro:
[image]

I may have paired devices, but bluetooth by default is turned off on my computers. Now that I think about it, this HC-06 must be the bluetooth module I experimented with (and I definitely used it on the iMac and the MacBook).

With Arduino connected, the following ports are listed:
[image]

Without Arduino:
[image]

edutec, 2016-02-02:
_From @maprzybylla on October 29, 2014 14:30_

That's it! I deleted the paired device from the list and now I can connect!

edutec, 2016-02-02:
_From @elaval on October 29, 2014 14:35_

Good!   I am glad we found the cause.   No we must see how to avoid this to happen again when user do have a paired bluetooth device.  What kind of device was paired? (One with serial communication bridged over bluetooth?)

@bromagosa I have never worked with Arduinos connected over bluetooth.  Can you give us some light on this?  Is there any way to distinguish a BT serialport used for an Arduino and one used for other devices?

edutec, 2016-02-02:
_From @elaval on October 29, 2014 14:37_

I guess we should rename the issue to something related to "Communication problems with Arduino when there are other bluetooth devices" and leave it open until we find a solution.

edutec, 2016-02-02:
_From @maprzybylla on October 29, 2014 14:41_

It's the HC-06 Bluetooth module and I used it to connect with Arduino. Description: http://www.hobbyist.co.nz/?q=bluetooth-module-device

edutec, 2016-02-02:
_From @bromagosa on October 29, 2014 15:23_

We've used this module in Citilab as well, but we never detected this
issue...

Anyways, this is not going to be easy! Maybe catching this error somehow?

I'll check it out tomorrow!

bromagosa, 2016-04-06:
We've just been testing this on a MacOSX with the latest Snap4Arduino version and it seems to be working now. Even when there's a paired Bluetooth device, we can successfully connect to an Arduino board by USB. I guess this was magically fixed when switching over to the chrome.serial API.

@maprzybylla, can you confirm whether it's okay in your side too? :)

maprzybylla, 2016-04-09:
Yes, now it shows me a list of all available ports including the BT device (even though it's disconnected) and I can just choose the right one. Great magical fixing! :)

maprzybylla, 2016-04-09:
Just an additional remark: if I try to connect to the BT-device attached to Arduino Uno (which as a user I'd assume is possible given the drop-dwon menu) I get this message: 
Could not talk to Arduino in port /dev/cu.HC-06-DevB
Check if firmata is loaded.

bromagosa, 2016-04-12:
Okay, that's a new bug it seems. I'm closing this one and filing in a new one.

Thanks! :)

closed 2016-04-12 by bromagosa