bromagosa/Snap4Arduino #28: New language files (Ukrainian and Russian)

SyavsoN, 2016-03-02:
Hi there!

Could you add two new language files (Ukrainian and Russian) to repository?

**Ukrainian** - [https://github.com/SyavsoN/Snap4Arduino/blob/master/snap/s4a/s4a-lang-uk.js](url)

**Russian** - [https://github.com/SyavsoN/Snap4Arduino/blob/master/snap/s4a/s4a-lang-ru.js](url)

Bets regards,
Yaroslav

bromagosa, 2016-03-03:
Thanks a lot! I'm integrating these in the next release! :)

SyavsoN, 2016-03-03:
Dear Bernat,

Could you tell me if you can integrate Ukrainian language for Snap, if I make the translation?

bromagosa, 2016-03-04:
Hi Yaroslav,

it'd be easier if you just send a pull request straight to https://github.com/jmoenig/Snap--Build-Your-Own-Blocks :)

SyavsoN, 2016-03-04:
Hi Bernat,

Thank you for your suggestion. I'm new to GitHub and version control, so your comment was very helpful, thank you!

commit 0d9627f3 (2016-03-16): closes #28
closed 2016-03-16 by bromagosa