bromagosa/Snap4Arduino #30: Cannot play mp3.

levindu, 2016-03-24:
I've tried the latest release in Arch Linux 64bits and Windows XP. Both cannot play mp3 sound.

bromagosa, 2016-03-24:
Can you try the same thing in regular Snap?

levindu, 2016-03-25:
I tried the regula Snap ( http://snap.berkeley.edu/snapsource/snap.html ), which can play the MP3 files both builtin or dropped to it.

edutec, 2016-04-22:
No problems here with version 1.1.2 in Debian stable. I'll try in Windows
ASAP.

Best regards,

-- 
*Víctor Casado*
Equip de recerca (Edutec)
victorct@citilab.eu

FUNDACIÓ PEL FOMENT DE LA
SOCIETAT DEL CONEIXEMENT
CORNELLÀ

Citilab Cornellà | http://citilab.eu | Pl. Can Suris s/n | 08940 Cornellà
de Llobregat | Barcelona

En compliment de la Llei Orgànica de Protecció de Dades de Caràcter
Personal (LOPD), l’informem que les seves dades de contacte han estat
incorporades en un fitxer titularitat de FUNDACIÓ PEL FOMENT DE LA SOCIETAT
DEL CONEIXEMENT (CITILAB). La finalitat d’aquest fitxer és la de mantenir
la relació amb els usuaris dels diferents oferts, inscripció als cursos i a
les activitats organitzades, facilitar la tramitació d’informació i
notificacions comercials i administratives, i en el cas de que vostè ho
hagi sol·licitat, l’enviament de publicitat relativa als serveis oferts.
FUNDACIÓ PEL FOMENT DE LA SOCIETAT DEL CONEIXEMENT (CITILAB) assegura la
confidencialitat de les dades aportades i garanteix que, en cap cas, seran
cedides per cap altre ús sense que hi hagi un consentiment previ i exprés
per part seva.

Per exercir els drets d’accés, rectificació, cancel·lació i oposició
previstos en la llei, pot adreçar-se mitjançant correu postal a FUNDACIÓ
PEL FOMENT DE LA SOCIETAT DEL CONEIXEMENT (CITILAB), Ref. Protecció de
Dades, Pl. Can Suris S/N, 08940, Cornellà de Llobregat, Barcelona, o bé
exercir els drets assenyalats mitjançant l’enviament de la seva petició al
correu electrònic avislegal@citilab.eu, indicant en l’assumpte del correu
“Ref. Protecció de Dades” (acreditant degudament la seva identitat).

bromagosa, 2016-07-04:
We can't fix this issue for now: https://github.com/nwjs/nw.js/issues/4492#issuecomment-192047570

Software patents... :(

bromagosa, 2016-07-07:
@levindu for now, the recommended way to go is to use a free format such as OGG.

levindu, 2016-07-08:
@bromagosa  Thanks for the effort.

closed 2016-07-08 by levindu