bromagosa/Snap4Arduino #32: Poor Bluetooth connection

chrisclaynz, 2016-04-04:
Hey all!

I've been working on a little robot with x2 DC motors connected via bluetooth. I've can have it working just fine as a remote controlled car but as soon as I pop a few IR sensors and try to get it following lines the bluetooth signal is really unstable. 

Has anyone else had unstable connections with Bluetooth... I'm pulling my hair out as I'm feeling I've tried every hardware issue so wonder if it's related to Snap4arduino..?

Help please :/

bromagosa, 2016-04-13:
@chrisclaynz, have you taken a look at whether it's just a power supply issue? We've been playing with BT and servomotors for about two hours straight now, and nothing went wrong yet.

We've tried it both in MacOSX and Windows. I can't get BT to work on my Gnu/Linux machine, but I'm afraid that's just my fault...

chrisclaynz, 2016-04-16:
Hey guys.

I think you are right and really sorry for bothering you about it. After
weeks of different battery combinations I found a way to ensure no power
dips and a really secure bt connection.

It was weird though as the issues I was having seemed less common with the
older version of s4a...

Anyway- I'm really sorry for wasting your time with this.

I'm looking forward to contributing more in the future.

Chris

awangenh, 2016-04-16:
Chris,

your difficulties could be due to communications speed. Bluetooth 
modules normally come factory set to operate at a speed of 38400 bps or 
9600 bps, whereas Firmata and FirmataPlus require that the module 
operates at 57600 bps.

When connecting a BT module in Windows, it set the speed automatically 
(MacOS seems to do the same thing). Linux does not do this, it uses the 
speed preconfigured in the BT module.

Here's a set of instructions on how to program your BT module to always 
use the 57600 bps speed:

http://www.computacaonaescola.ufsc.br/?page_id=43&lang=en

-- 

"It is difficult to get a man to understanding something
when his salary depends upon his not understanding it."

Upton Sinclair

=================================================================
Prof. Dr. rer.nat. Aldo von Wangenheim

Executive Director
National Research Institute on Digital Convergence - INCoD
Universidade Federal de Santa Catarina - UFSC
88049-200 Florianopolis - S.C.
Brazil

Phone: +55-48-3721-4715 (INCoD/LAPIX)
        +55-48-3721-4714 (Telemedicine)

E-Mail: awangenh at: inf.ufsc.br
Skype:  awangenh

Personal:    http://www.inf.ufsc.br/~awangenh
INCoD:	     http://www.incod.ufsc.br/

Coordinator:
         Cyclops Group Brazil
         http://www.cyclops.ufsc.br/

         Image Processing and Computer Graphics Lab
         http://www.lapix.ufsc.br/

         Telemedicine Lab/UFSC University Hospital
         http://www.telemedicina.ufsc.br/

=================================================================

awangenh, 2016-04-16:
There's also a Spanish version of this page: http://www.computacaonaescola.ufsc.br/?page_id=43&lang=es

bromagosa, 2016-04-18:
@chrisclaynz no worries, it helped us too! :)

closed 2016-04-18 by bromagosa

chrisclaynz, 2016-04-19:
Hey Aldo,

Thanks for your suggestions. I had already done this but found that the
power required by DC motors results in dips which cause the bt mod to reset
and connection to be lost. I have remedied this by providing the bt mod
with it's own power supply.

Chris

awangenh, 2016-04-19:
Thank you for your feedback.
Brushed motors produce a lot of tension variations and really need an 
independent power supply. We also  discovered this  the hard way ... :-(

-- 

"It is difficult to get a man to understanding something
when his salary depends upon his not understanding it."

Upton Sinclair

=================================================================
Prof. Dr. rer.nat. Aldo von Wangenheim

Executive Director
National Research Institute on Digital Convergence - INCoD
Universidade Federal de Santa Catarina - UFSC
88049-200 Florianopolis - S.C.
Brazil

Phone: +55-48-3721-4715 (INCoD/LAPIX)
        +55-48-3721-4714 (Telemedicine)

E-Mail: awangenh at: inf.ufsc.br
Skype:  awangenh

Personal:    http://www.inf.ufsc.br/~awangenh
INCoD:	     http://www.incod.ufsc.br/

Coordinator:
         Cyclops Group Brazil
         http://www.cyclops.ufsc.br/

         Image Processing and Computer Graphics Lab
         http://www.lapix.ufsc.br/

         Telemedicine Lab/UFSC University Hospital
         http://www.telemedicina.ufsc.br/

=================================================================