bromagosa/Snap4Arduino #34: Bluetooth apparently not working on Gnu/Linux or Windows

bromagosa, 2016-04-06:


chrisclaynz, 2016-04-06:
Hey guys!

I'm wondering this is related to my earlier inquiries about poor Bluetooth connections?

I will try the servo suggestion but this really won't be fit for purpose in the long term... I am really keen to Get this working- the snap4arduino project is so cool. 

Cheers,

Chris

edumaker, 2016-04-07:
i am a teacher in china and i write a lot of books in china.s4a is my best love software.and i met snap team at scratch@mit2012.now i noticed you have a project using scratch to build 3Dmodel.but i cannot find it now and i hope this projects works  very well.because i want to intruds it to chniese techers. hoping you have a good time eceryday and thanks for all S4A team

ontouchstart, 2016-04-07:
@bromagosa Hi Bernat,

I introduced @edumaker to GitHub community and this is his first post on GitHub. I know it is not the comment is off topic for this issue itself. Can you give him a guide on proper communication channel between Chinese user community and your team? I **guess** his specific question is about exporting 3D models to [STL format]( https://help.github.com/articles/3d-file-viewer/).

Thanks for the excellent work :+1: 

Sam @ontouchstart

bromagosa, 2016-04-07:
Hi!

No problem, for further enquiries please use our team email: edutec _at_ e-citilab _dot_ eu

I believe you're talking about Beetle Blocks, you can find it here: http://beetleblocks.com

Are you going to be in Scratch@MIT this year? If so, looking forward to meet you there! :)

edumaker, 2016-04-08:
i think i will.do you have some news about scratch@mit2016?

bromagosa, 2016-04-13:
So, Bluetooth seems to be working just fine in Windows and MacOSX. And I truly believe my problems witih Gnu/Linux are system-related.

I'm closing this for now. What I agree on is that the error messages should be a little bit more informative than [Object object], which I'll address in a new issue.

closed 2016-04-13 by bromagosa

DavidGasku, 2017-02-27:
Hi, I think I have the same problem you had. s4a says [Object object], and throws `[17022:17045:0227/212403:ERROR:serial_io_handler.cc(152)] Failed to open serial port: FILE_ERROR_IN_USE` in the terminal. I'm using ubuntu 16.10.
Did you finally solve the problem?

jguille2, 2017-02-28:
Hi @DavidGasku ,

If your error is "in_use" (or busy), another service in your SO is occupying this connection.

I've seen in some distros that ModemManagerService is always running and using BT connections.

You can try:
`
    sudo systemctl stop ModemManager.service
`
and if this is the problem, you can disable this service permanently.
`
    sudo systemctl disable ModemManager.service
`

I Hope this is useful.

Joan