bromagosa/Snap4Arduino #36: Bluetooth port shows up in MacOSX but cannot be used

bromagosa, 2016-04-12:
@maprzybylla wrote:
> if I try to connect to the BT-device attached to Arduino Uno (which as a user I'd assume is possible given the drop-dwon menu) I get this message: 
> Could not talk to Arduino in port /dev/cu.HC-06-DevB
> Check if firmata is loaded.
> 
[image]

> 
> And one more thing: if I then click ok and try to connect again, nothing happens :(

bromagosa, 2016-07-05:
I'm pretty sure this one is fixed already, can someone with a MacOSX confirm it, please?

tiagofer, 2016-07-25:
Hi, I'm using 1.1.6-beta-rc version on windows 10 and I have the same problems with you. I tested the FirmataPlus and StandardFirmata but do not work.

hakanatas, 2016-07-30:
I have the same problem with MacOSX and also win10

eglix, 2016-07-30:
Hi, i tried connect with standardfirmata and modified firmata(myself) but i did not get result.

bromagosa, 2017-02-14:
Many users are successfully connecting Snap4Arduino over Bluetooth in MacOSX, Windows and GNU/Linux. I suspect there's something wrong in your particular setups...

Are you sure you're following [the instructions](https://github.com/bromagosa/Snap4Arduino/wiki/Autonomy-via-Blueetooth)?

closed 2017-02-14 by bromagosa