bromagosa/Snap4Arduino #43: Servo issues

jguille2, 2016-05-26:
## 4 comments/suggestions about Servos

### 1. I suggest to change angle mapping
  - Default mapping is 0-180 degrees to 544-2400 microseconds
  - Then 90º are 1472µs. Best value is 1500µs.
  - Proposed change:
    - */snap/s4a/threads.js*, after [line 69](https://github.com/edutec/Snap4Arduino/blob/master/snap/s4a/threads.js#L69)
    -  Add ```board.servoConfig(pin,600,2400);```
  - Then 90º will be 1500 µs

### 2. Fix same speed for clockwise and counter-clockwise operation
  - Regardless the mapping change proposed for *angles* operation, 1500 is the right value for stopped state.
  - Then, clockwise and counter-clockwise must have the same increment (the same speed).
  - Proposed change:
    - */snap/s4a/threads.js*, [line 77](https://github.com/edutec/Snap4Arduino/blob/master/snap/s4a/threads.js#L77)
    - We can **change 1700 to 1800** value

### 3. Add block info to change servo speed
  - We can add more options to block (fast, slow...)
  -  But maybe it's easier add info like *angle* option: ``'angle (0-180)' : 90,``
  - Proposed change:
    - */snap/s4a/blocks.js*, lines [12-16](https://github.com/edutec/Snap4Arduino/blob/master/snap/s4a/blocks.js#L12-16)
    ```
    'angle (0-180)' : 90,
    'stopped' : ['stopped'], 
    'clockwise (1500-1000)' : ['clockwise'],
    'counter-clockwise (1500-2000)' : ['counter-clockwise'],
    'disconnected' : ['disconnected']
    ```
  - Note: I've tested that values greater than 2000 and smaller than 1000 don't increase speed (almost in some servos)

### 4. Disconnecting servos does not enable pwm in 9 and 10 pins.

  - Disconnect must detach servo.
  - Proposed change:
    - *[/snap/s4a/threads.js](https://github.com/edutec/Snap4Arduino/blob/master/snap/s4a/threads.js#L83)*
    - Change
    ```this.digitalWrite(pin, false);```
    - to
    ```board.pinMode(pin, board.MODES.OUTPUT);```
  - Unfortunately, this change does'nt fix de problem, because there is a [bug](https://github.com/arduino/Arduino/issues/3860) in Servos library. We can see the [missing code](https://github.com/arduino/Arduino/blob/master/libraries/Servo/src/avr/Servo.cpp#L205). We must wait. We can implement a Firmata hack to enable timer ending (see Arduino issue) but that does not run in every board (manage only one 16 bit timer).

Thanks!!

Joan

jguille2, 2016-05-27:
@bromagosa 
I'll make a PR with the changes... but I wait feedback/discussion about this.

bromagosa, 2016-05-27:
Hi @jguille2!

I'm perfectly fine with **1**, **2** and **4**, but point **3** is a bit controversial.

In the beginning I used to show the actual values when you selected ``clockwise`` or whatever other label, but some users complained then the code was unreadable (``set servo to 1500`` is not a human-friendly instruction). That's why I decided to just map the labels internally.

However, you can already control speed with the current blocks if you write values in the servo block. My suggestion would be a compromise between what you want and what I want. What about an extra option that says ``custom rotation (1000-2000)`` and makes clear to users they can experiment with values in that range to get different rotation speeds and directions?

jguille2, 2016-05-27:
Hi @bromagosa 

But my proposal is not like ``set servo to 1500``!!

In the dropdown options we'll see:
[image]

I don't change values. Blocks keep their human-friendly text (stopped, clockwise, counter-clockwise, disconnected or a number). The change is only in the dropdown.

If you choose 'counter-clockwise (1500-2000)', the block will show ``set servo to counter-clockwise`` (just like now), and the 'real value' will be 1800 (but it's not showed). But you've seen before your options (1500-2000) in the dropdown. It's only info.

[image]

Of course, if you decide to use speed control, then you'll use numbers (like ``set servo to 1650``). But basically, we continous working with 'basic strings' with the info needed to allow to implement more control.

bromagosa, 2016-05-27:
Oh I see! Then it's all good! :)

jguille2, 2016-05-29:
Closed with #46 PR

closed 2016-05-29 by jguille2