bromagosa/Snap4Arduino #50: Bluetooth Connectivity

TheDiscovery, 2016-06-15:
Hi,

Just posting a quick question. I want to remotely program my Arduino UNO. Can I simply connect a bluetooth module via the Arduino's tx and rx pins, pair to the module and use Snap4Andriod with the appropriate COM port or do I need to modify the Firmata to exclude pins 1&2 from the digital inputs? Sorry for the newbie question.

bromagosa, 2016-06-15:
Your first hunch is 100% correct, that's exactly what you need to do. No need to modify Firmata, but remember to set the BT module at the right speed for Firmata to communicate with the board (115200).

closed 2016-06-15 by bromagosa

edutec, 2016-06-15:
Watch out, the speed for Snap4Arduino is 57600!

Regards.

-- 

*José García*Coordinador àrea formació
+34 93 551 06 75 (ext.120)
jgarcia@citilab.eu

FUNDACIÓ PEL FOMENT DE LA

SOCIETAT DEL CONEIXEMENT

CORNELLÀ

Citilab Cornellà | http://citilab.eu | Pl. Can Suris s/n | 08940 Cornellà
de Llobregat | Barcelona

En compliment de la Llei Orgànica de Protecció de Dades de Caràcter
Personal (LOPD), l’informem que les seves dades de contacte han estat
incorporades en un fitxer titularitat de FUNDACIÓ PEL FOMENT DE LA SOCIETAT
DEL CONEIXEMENT (CITILAB). La finalitat d’aquest fitxer és la de mantenir
la relació amb els usuaris dels diferents oferts, inscripció als cursos i a
les activitats organitzades, facilitar la tramitació d’informació i
notificacions comercials i administratives, i en el cas de que vostè ho
hagi sol·licitat, l’enviament de publicitat relativa als serveis oferts.
FUNDACIÓ PEL FOMENT DE LA SOCIETAT DEL CONEIXEMENT (CITILAB) assegura la
confidencialitat de les dades aportades i garanteix que, en cap cas, seran
cedides per cap altre ús sense que hi hagi un consentiment previ i exprés
per part seva.

Per exercir els drets d’accés, rectificació, cancel·lació i oposició
previstos en la llei, pot adreçar-se mitjançant correu postal a FUNDACIÓ
PEL FOMENT DE LA SOCIETAT DEL CONEIXEMENT (CITILAB), Ref. Protecció de
Dades, Pl. Can Suris S/N, 08940, Cornellà de Llobregat, Barcelona, o bé
exercir els drets assenyalats mitjançant l’enviament de la seva petició al
correu electrònic avislegal@citilab.eu, indicant en l’assumpte del correu
“Ref. Protecció de Dades” (acreditant degudament la seva identitat).

bromagosa, 2016-06-15:
Ouch! That's what I meant, sorry! :sweat_smile: