bromagosa/Snap4Arduino #69: No message if cable is disconnected

toreonify, 2016-08-24:
I noticed that in code there is an condition when cable gets unplugged. But in current version it doesn't show that message.

```
if (this.gotUnplugged & !silent) {
        ide.inform(
                this.owner.name,
                localize('Board was disconnected from port\n') 
                + portName 
                + '\n\nIt seems that someone pulled the cable!');
        this.gotUnplugged = false;
    } else if (!silent) {
        ide.inform(this.owner.name, localize('Board was disconnected from port\n') + portName);
    }
```

So I debugged that handler and found what's wrong with that. The handlers disconnectHandler and closeHandler connected to Firmata but code in handlers accesses Arduino object. When I log "this" variable in those handlers, they return SerialPort object.

```
myself.board.sp.on('disconnect', myself.disconnectHandler);
myself.board.sp.on('close', myself.closeHandler);
myself.board.sp.on('error', myself.errorHandler);
```
In closeHandler interval doesn't get cleared, because interval ID defined in sprite's Arduino object.

```
Arduino.prototype.closeHandler = function (silent) {

    var portName = 'unknown';

    clearInterval(this.keepAliveIntervalID); // this return SerialPort

    if (this.board) { // no board property in SerialPort
        portName = this.board.sp.path;

        this.board.sp.removeAllListeners();
        this.board.sp = undefined;

        this.board = undefined;
    };
```

Tested on OS X 10.11.6 with NW.js v0.15.4 and latest Firmata node module. If it wouldn't replicate on your system, I will try clean install of S4A.

toreonify, 2016-08-24:
So, I found a some sort of way to fix it: add an Arduino object to SerialPort when connect is successful.

```
myself.board.sp.arduinoOwner = myself;
```

In handlers we will check what object is called it. Arduino object doesn't have a property referencing to itself, but SerialPort has a property referencing to  Arduino object.

```
var myself = this.arduinoOwner ? this.arduinoOwner : this;
```

For me – works well.

bromagosa, 2016-08-26:
You are right, this happened after refactoring Arduino functions into a separate object.

I'll fix this ASAP, thank you! :)

commit 9ced0fd5 (2016-08-26): fixes #69, plus adds SV translation by Ove
closed 2016-08-26 by bromagosa