bromagosa/Snap4Arduino #70: Only a question - please

buenow, 2016-08-24:
Sorry for my question may seem obvious. I'm newbie on github. I'm from Brazil and I´m not so good in english, excuse-me.  I tryed set up the bluetooth module (using pins 10/11 of arduino uno). I loaded firmata on it. Then I can´t use snap4arduino with bluetooth, after change pins 10/11 by 0/1 because every time i lose my conection with snap4arduino when I try to use both. I can´t read analogic pin 0 and use this data in the snap4arduino.  I spent some days trying to solve this... Can someone help me please? In details, of course, if possible.  Your help will be very important for us. Thank you very much. (I am teacher of kids and i am trying to help they usando arduino).

jguille2, 2016-08-25:
Hi @buenow,
I try to explain...
-We can't use pins 10/11 with StandardFirmata. We have not extra serial libraries.
-We can use pins 0/1 (digital). Then BT is replacing USB connector. And I'll connect the Sprite with the board by this connnection (BT appears like a serial device). We must set BT baud to 57600 before.
-Then we have connected Sn4A with Arduino without a cord.
-If connection is ok, we can do the same actions that we do when we are connected by USB. But not more. We can't read 0/1 digital pins, because we are using to communicate PC and board.

Maybe this was not the problem ... Give us more details and we will try to find solutions.

Joan

buenow, 2016-08-25:
Hi Joan. Thank you .  You was very gentile in this answer (very quickly).

When I say i used pins 10/11: I had used that pins to set configurations of
BT.
After that, I changed pins 10 by 0 and 11 by 1 (according the catalan
article that follows).

By other side, with BT configured, I have no success to use Sn4A.

When I tryed to use these pins of BT connected to 0 and 1, I lost my
connection with Sn4A and Arduino.

What I did not understand is what I need to do to use BT with Sn4A? What
kind of object in Sn4A I need to use to connect my Sn4A with BT?

I followed these steps that was showed in that article in catalan, about BT
and a car.

https://docs.google.com/document/d/1FiSmQ-f3wEcA3MtlX67GMQiVmWQ-fFYuSLsNPFSfu1A/edit

I dont understand steps 2 and 3 about how to stablish conection between BT
and Sn4A.
(3 - specifically - Connexió: necessitem connectar abans d’obrir el port
des de l’S4A o l’Snap4Arduino. Una vegada connectat, el LED del mòdul ha
d’estar fixe.) .

We are trying to do this car for our special kids -
https://www.youtube.com/watch?v=x4c7437ZxUUhttps://www.youtube.com/watch?v=x4c7437ZxUU

I was trying this tutorial: https://www.youtube.com/watch?v=hr23HfC3OWQ.

We think that you are very busy people. We are very grateful for your help.

Thank you very much.

Wilson Bueno and our students.

jguille2, 2016-08-25:
Hi Wilson,
I try to explain this...

-Step 1 is for BT configuration. If you did it, it's ok and I can forget this. The board is only a tool (you can do this with another board).

-Step 2 is for board (Arduino) configuration. Snap4Arduino is not a tool to program Arduino. The program runs on the PC (Snap is running). Then, we need StandardFirmata sketch into the board. And then, arduino does whatever Snap tell it. 
  - You need Arduino IDE. In examples, you have Firmata->StandardFirmata scketch.
  - Step 2 is needed to work with Snap4Arduino. You can do this and play with the board without the BlueTooth. I suggest you to try it. After Step 2, plug the USB cord (PC to Arduino), use the Connect button in Snap and you can begin to play. Test this (without the BT).

-With steps 1 and 2 you have all configured. You want to use BT instead of the USB cord.
-Then, with Arduino not connected, without the USB cord, you must make the connection by BT. Is the Step 3. You must pair the BT and connect.

Details are different in Windows and Linux. Try this and report us the details if you have a problem...

Continue...

Joan

buenow, 2016-08-25:
Thank you again Joan.

I will test.  My arduino is OK, with firmata running on it and run without
BT.
I did not try to use BT without cord... Oh I did not think about this...
rsss.

Maybe because i am doing the program in Sn4A and testing simultaneously.

I will verify your tips.

Thank you again Joan.

Wilson

:

buenow, 2016-08-25:
Joan.  Only one more thing...

I thought the program made in sn4a was not tranfered to the microcontroller
memory, depending in this case of the cord to stablish comunication between
sn4a and arduino board.

There are any docs or book that let me to know sn4a in more details?  What
kind of lecture do you recomend for my study about sn4a? I am electronic
technician with some experience with programs  and now I work with special
kids (children with cognitive disabilities).  This car that we are doing
with your help will be very usefull for they.

Thank you again.

Wilson

2016-08-25 11:27 GMT-03:00 wilson bueno <contate.wilson@gmail.com>:

buenow, 2016-09-27:
Thank you Joan.

Finally evething is ok.
Your help was great.

Wilson

closed 2016-09-27 by bromagosa