bromagosa/Snap4Arduino #71: How to add new block permanently in s4a

Sarfaraazahmed, 2016-09-09:
(http://blog.s4a.cat/2015/03/13/Extending-Firmata-for-Snap4Arduino.html ) I went through this step to create new block but once i close and open s4a it disappear. Each time i have to create new then i export that block to reuse but that is painful activity each time i have to import. How to fix permanently in s4a  so that it stays back

jguille2, 2016-09-09:
Hi,

The only way to fix a block into the palette is a hack of the software.

But I think it's not necessary. When you started a project, you'll import different blocks (custom blocks, tools, libraries...) often. It's the way to use past work and not repeat it. And after this, you must load your project (with all these blocks) to continue...
Joan

closed 2016-10-05 by bromagosa

Robert439, 2017-11-21:
Hi, I need a version of sna4arduino with some modifications, please contact  me @bromagosa 
Best regards.

jguille2, 2017-11-21:
Hi @Robert439 ,
Right now Bernat is on vacation. If you want some help, we can continue into this issue...

Anyway, I insist that you can open directly templates. You can open it directly by URL into the online version (opening a project stored and shared in the cloud)  or by a DesktopShortcut in the desktop version (opening a local xml).