bromagosa/Snap4Arduino #74: Premature leaving page dialog

stirlingstout, 2016-09-26:
Snap4Arduino! 0.1.5-beta

I've got a project which uses a Python server to implement a file server, so Snap! or Snap4Arduino can write/rread data (e.g., logged from an Arduino) to a file.

I've just been trying to implement a missing block (see the pictures) and when I try to Apply the change to the blocks in the first picture I get asked if I want to leave the page. If I take the call block out then Apply works OK. I've net yet confirmed if this happens in Snap!

[image]

[image]
[image]
[image]
[image]

bromagosa, 2016-09-27:
Hi John!

Obvious question (sorry), have you tried using the latest Snap4Arduino version?

I assume you're using the desktop flavor, right? If that's the fact, you could actually implement file system operations without the need of an intermediary server, here's an example:

[image]

Still, I'm not sure what can be causing the leave page dialog to show up. Can you attach the project XML here so I can take a look?

Thanks! :)

stirlingstout, 2016-09-27:
Bernat

I’d just downloaded the version that I thought was the latest from the site (1.0.6 Beta). I haven’t tried it with the web version yet.

I didn’t know about the require(‘fs’) technique, which is great, but I wanted a solution for ‘normal’ Snap! to run in the browser.

I’ll have another go at reproducing the problem and  send you the project XML.

John

[image]<https://cloud.githubusercontent.com/assets/1016697/18862527/007bf210-848d-11e6-8cf7-ee86bc26d829.png>

This email was Virus checked by KGV Sophos UTM. itservices@kgv.ac.uk<mailto:itservices@kgv.ac.uk>

[King George V Logo]
John Stout
Management Information Systems Coordinator
Tel: 01704 530601
E-Mail: JSS@kgv.ac.uk
Web: www.kgv.ac.uk

Information contained in this e-mail is intended for the use of the addressee only and is confidential.  Any dissemination, distribution, copying or use of this communication without prior permission of the addressee is strictly prohibited.

Every effort has been made to ensure that any attachment to this mail does not contain virus. King George V College has taken every reasonable precaution to minimise this risk, neither it nor the sender can accept liability for any damage which you sustain as a result of software viruses. You should carry out your own virus checks before opening any attachments.

[Think Before You Print]

jguille2, 2016-09-27:
Last version is 1.1.6 beta-rc3 released on 19/08/2016.

Current site is http://snap4arduino.org/

stirlingstout, 2016-09-27:
The version I'm running I downloaded from the site http://snap4arduino.org/ yesterday, using the Microsoft Windows (64 bits) link, but clicking on the logo when running Snap4Arduino and  selecting About Snap4Arduino... shows 0.1.5-beta.

John

bromagosa, 2016-09-27:
Ouch! That's my fault! Thanks for reporting this!

stirlingstout, 2016-09-27:
?Oh, and the title bar shows Snap4Arduino 1.0.6-beta.

John

stirlingstout, 2016-09-27:
Bernat

Does that mean you can reproduce the problem? Or have found the solution? If so I won't send you the xml.

John

bromagosa, 2016-09-27:
Nope, that just means I messed up the version reporting in the About dialog, but I still need the XML to test what's going on with your project :)

stirlingstout, 2016-09-27:
Here's the project. If I edit the block called **(table from file called (filename))** I can Apply it as it currently, but if I drag the **map** block that's on the scripting area into the **[report []]** and do Apply I get asked if I want to leave the page.

I've put my SnapFiles on Github at stirlingstout/SnapFiles. So far it works really nicely with an Arduino and Snap4Arduino, but it would be nice to have the tables feature from the current Snap! (and keyboard editing) to display data logging files.  

[PrematureLeavingPage.zip](https://github.com/edutec/Snap4Arduino/files/495287/PrematureLeavingPage.zip)

bromagosa, 2016-09-27:
Hmm... I see, it does work in my PC on the latest dev version. Can you try with the version I'm linking?

http://snap4arduino.org/downloads/test/linux32.zip
http://snap4arduino.org/downloads/test/linux64.zip
http://snap4arduino.org/downloads/test/osx64.zip
http://snap4arduino.org/downloads/test/win32.zip
http://snap4arduino.org/downloads/test/win64.zip

stirlingstout, 2016-09-27:
I tried the Win64 version and that doesn't give the error, but when I import the project it tells me it was made with a different version of Snap!

bromagosa, 2016-09-27:
Yep, that's a known issue I'll resolve before publishing the next release.

So it seems it's fixed in my dev branch, I'll close this issue when I release it :)

Thanks for your patience!

stirlingstout, 2016-09-27:
Which are in the beta version you pointed me to!

bromagosa, 2016-09-27:
yep :)

I'm just waiting to fix an annoying bug to publish it, we'll hopefully fix it tomorrow with @jguille2...

bromagosa, 2016-10-06:
About to release 1.2, so I'm closing this.

closed 2016-10-06 by bromagosa