bromagosa/Snap4Arduino #77: Where is the source for the "Demos" or "Examples" ?

bradparks, 2016-10-04:
Hey! This looks great - I find I work best if working from Examples, and I see the great demos on the homepage, but I can't see the source in the github repo, or in the StandardFirmata repos either.

For example, I don't see the Etch a Sketch one there. Are there any examples that come with this?

Thanks,

Brad

bromagosa, 2016-10-05:
Hi Brad,

we know this is a problem, and we should definitely try to allocate some resources into putting a bunch of examples together.

In some cases it's just a matter of exporting the project and linking to it. I'll try to do these first.

Sorry about that.

bradparks, 2016-10-05:
No worries... thanks a ton for this - i see that some of them are available on the links of the YouTube videos, so I'll follow through there... Thanks again!

bromagosa, 2016-10-05:
Here's some examples I had laying around in my computer:

https://github.com/edutec/Snap4Arduino/tree/master/examples

Hope they help you get started! :)

closed 2016-10-05 by bromagosa