bromagosa/Snap4Arduino #80: Error while building on ubuntu machine

najiJazzar, 2016-10-06:
I'm trying to build the repo as descriped. I'm using a 32-bit ubuntu-16-04 machine. So I prepared the installation using the --desktop option and built it using the --platform=desktop option. Then from the releases directory I tried to lauch the Snap4Arduino but the following error appeared :
./Snap4Arduino.desktop: line 1: [Desktop: command not found
./Snap4Arduino.desktop: line 5: -c: command not found
./Snap4Arduino.desktop: line 8: Snap!: command not found

Am I doing something wrong ?

bromagosa, 2016-10-06:
The executable is `Snap4Arduino`, not `Snap4Arduino.desktop`. The latter is a launcher for Nautilus and other file managers (I think, I'm not a GUI user...)

So, you should either do:

```
./build --platform=desktop/gnu/32 --run
```

And the build script will automatically launch Snap4Arduino after building it, or:

```
cd releases/desktop/gnu/32
./Snap4Arduino
```

Or, after having successfully built Snap4Arduino, you can always call the build script like this:

```
./build --rerun
```

Which will look for the appropriate build for your system and launch it.

Hope it works! :)

--

@jguille2 I knew the desktop file would bring issues! ;)

bromagosa, 2016-10-06:
@najiJazzar I've renamed the GNU/Linux executable, now it's called `run`.

You can still use all these ways to run Snap4Arduino by using the builder, except that the second one now looks like this:
```
cd releases/desktop/gnu/32
./run
```

closed 2016-10-06 by bromagosa

najiJazzar, 2016-10-07:
@bromagosa 
The problem is that the executable (Snap4Arduino or run )isn't generated.  The --run option gives me the following error:

**_ERROR: Auto-running is not supported for your current architecture or operating Your system reports this is an i686 machine. This script can only auto-run for x86_64 and i386 under GNU/Linux.
In other systems, please build Snap4Arduino by using
./build --platform=TARGET
and then manually navigate to the releases directory._**

And the result of `ls releases/desktop/gnu/32 ` is:
**icons  launcher.sh  LICENSE  Snap4Arduino.desktop**

I don't know why "run" doesn't appear !

bromagosa, 2016-10-07:
I see, that's the same issue @jguille2 had a while ago, and IRC it had to do with the NodeJS version he had installed in his machine. Can you paste here what's the output of runnning `node --version` and `npm --version`?

Thanks!

najiJazzar, 2016-10-07:
It seems like this is the problem, that's embarrassing, I didn't have the NodeJS
I'll redo the steps now hopefully now it will work.
Thanks a lot

bromagosa, 2016-10-07:
Nah, that's my fault. I should have documented the prerequisites... let me get to that :)

najiJazzar, 2016-10-07:
@bromagosa 
Am sorry, but also with downloading the NodeJS and the other things you've mentioned in the new description, it still not working with me :(
same problem, there is no executable.
here is the versions you've requested before :

`node --version` = v4.2.6
`npm --version` =3.5.2

bromagosa, 2016-10-07:
Your NodeJS version is too old I believe, mine is `v6.1.0`.

Try upgrading it, you can follow the instructions from https://nodejs.org/en/download/package-manager/ to install v6.

Hope it works :)

najiJazzar, 2016-10-07:
now I have v6.7.0
and the executable is still missing. any thoughts  ?
Sorry for bothering you :\

bromagosa, 2016-10-07:
No worries, that's what I'm here for :)

Did you run `./prepare --all` again? Some packages will need to be rebuilt after installing the new Node version.

najiJazzar, 2016-10-07:
I removed the whole directiory, cloned it again.
so yes I did
but `./prepare --all  ` when it finishes it gives me this message :

**WARNING: Please make sure to place your Snap4Arduino.pem Chromium extension private key under environments/web/chromium
I cannot do this for you because, you know, it's a private key ;)**

I was ignoring this message since I think I don't need any Chromium-related things
I don't know if this creates a problem !

bromagosa, 2016-10-07:
Nope, this should not be a problem.

That's weird... what error message(s) does `./build --platform=desktop/gnu/32` give out?

najiJazzar, 2016-10-07:
no error messages
the last line printed is a green-colored `Done!`

bromagosa, 2016-10-07:
wow.. and still there's nothing under `releases/desktop/gnu/32`?

@jguille2 is this what was also happening to you? How did you fix it?

bromagosa, 2016-10-07:
Also, can you please run this and paste the contents of the `tmp` directory here?

`./build --platform=desktop/gnu/32 --keeptmp`

najiJazzar, 2016-10-07:
Unfortunenately yes, it still missing.

here is the `ls tmp` result: 
[image]

agpl.txt                   lang-pt.js
Backgrounds                lang-ro.js
blocks.js                  lang-ru.js
byob.js                    lang-si.js
click.wav                  lang-sv.js
cloud.js                   lang-ta.js
contributing to BYOB4.txt  lang-te.js
Costumes                   lang-tr.js
Examples                   lang-tw.js
favicon.ico                lang-zh.js
FileSaver.min.js           libraries
gui.js                     lists.js
help                       locale.js
history.txt                manifest.mf
index.html                 morphic.js
lang-ar.js                 morphic.txt
lang-bg.js                 node_modules
lang-bn.js                 objects.js
lang-ca.js                 optimizations.txt
lang-cs.js                 package.json
lang-de.js                 paint.js
lang-dk.js                 README.md
lang-el.js                 readme.txt
lang-eo.js                 s4a
lang-es.js                 s4a.ico
lang-et.js                 s4a_logo_sm.png
lang-fi.js                 sha512.js
lang-fr.js                 Snap! Connectivity Strategy.pdf
lang-hr.js                 snap_fast.html
lang-hu.js                 snap.html
lang-ia.js                 snap_logo_sm.png
lang-id.js                 Sounds
lang-it.js                 store.js
lang-ja_HIRA.js            tables.js
lang-ja.js                 threads.js
lang-kn.js                 tools.xml
lang-ko.js                 translating Snap.txt
lang-ml.js                 version
lang-nl.js                 widgets.js
lang-no.js                 xml.js
lang-pl.js                 ypr.js
lang-pt_BR.js

bromagosa, 2016-10-07:
Hmm... the tmp directory looks perfectly fine to me. It's exactly what should be there.

Let's break this down step by step. Here's my whole output, can you check if there are any differences to yours?

http://pastebin.com/raw/Fj9eVRAE

najiJazzar, 2016-10-07:
wow, there are files that aren't found !
here is what I get:

http://pastebin.com/raw/tVPBZGzx

bromagosa, 2016-10-07:
Now it starts to make sense. `nw-builder` is not installed, but that should have been handled by the `prepare` script. Can you paste the output of `./prepare --nwjs` please?

najiJazzar, 2016-10-07:
here it is:

naji@naji:~/Snap4Arduino$ ./prepare --nwjs
Pulling nwjs.io builder (nw-builder)
npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
/home/naji
└── nw-builder@3.1.0 

npm WARN enoent ENOENT: no such file or directory, open '/home/naji/package.json'
npm WARN naji No description
npm WARN naji No repository field.
npm WARN naji No README data
npm WARN naji No license field.
nwjs.io installed.

bromagosa, 2016-10-07:
What's under `environments/node_modules/nw-builder/bin/` ?

najiJazzar, 2016-10-07:
there is no such directory in mine
this directiory instead `/Snap4Arduino/environments/desktop/node_modules` have :
`async` `bindings` `browser-serialport` `debug` `es6-shim` `firmata` `minimist`  `ms`  `nan`  `optimist`  `sf` `wordwrap`

bromagosa, 2016-10-07:
What about in `environments/node_modules`?

najiJazzar, 2016-10-07:
`environments` only have
 `desktop` `embedded` `web`

bromagosa, 2016-10-07:
That is really weird, but what it seems is that npm has installed nwbuilder somewhere else in your system.

This will take a while, but can you run `find ~ -name nwbuild` ?

najiJazzar, 2016-10-07:
Here what it says :
/home/naji/node_modules/.bin/nwbuild
/home/naji/node_modules/nw-builder/bin/nwbuild

should I move them ?

bromagosa, 2016-10-07:
Yep! Just move the whole `node_modules` directory inside `environments`, and that should make `./build` work!

najiJazzar, 2016-10-07:
booooyah!!
finally it wooorks :D
thanks a lot really for your precious time :)

bromagosa, 2016-10-07:
Right on! There we go!

Now I should find out why was this happening in the first time though. I guess I could force `prepare` to install these packages in the working dir.

Thank you too for helping debug it! :)

commit 488ca167 (2016-10-06): Renamed Snap4Arduino executable in GNU/Linux to avoid confusion (closes #80)