bromagosa/Snap4Arduino #86: Mega, second hw serial port usage

microspace, 2016-10-18:
I've seen several questions about hw serial, but I could find definite answer, so,

1)Can I use other hw serial port of arduino mega in snap4a?

2) I can't see any serial read/write bricks in snap4a interface? Am I naive about something?

bromagosa, 2016-10-19:
Hi!

Do you mean I2C? If so, I'm still developing the blocks that let you use it, but here's a preliminary version that you can just drag and drop into Snap4Arduino to test them out :)

[i2c.zip](https://github.com/edutec/Snap4Arduino/files/538403/i2c.zip)

microspace, 2016-10-19:
Thank you, i2c will be useful, but I mean UART serial port, mega has 4 of them. I want to receive data from a device by UART 
Another question: is SPI communication supported?

bromagosa, 2016-10-19:
Oh I see, I've never actually played with it, but it seems Firmata [does support it](https://github.com/firmata/firmata.js/issues/95) since a while ago, so we should be able to easily add this functionality to Snap4Arduino.

Could you explain to me in a bit of detail how you want to use UART (devices involved, simple schematics)? I would want to replicate your use case and see if I can make it work myself.

SPI is not (yet) supported by Firmata, so there isn't much I can do about it for now...

microspace, 2016-10-19:
I want to receive data from Pixy Camera. http://cmucam.org/projects/cmucam5/wiki/Pixy_Serial_Protocol

Pixy can send data via uart, i2c, spi and even analog ports. Uart seams to me as most obvious choice. 
But I2c is also worth trying. 
From your answer I understood that I can receive data from Pixy. 
Pixy sends textual info about detected objects.
Now I'll investigate how to code a snap block to receive data from Pixy and parse it.
In the end I want to have a span block or a couple of variables that holds coordinates of detected objects.
Thank you for your support!

closed 2023-05-25 by jguille2