bromagosa/Snap4Arduino #93: Implerement "map" function

microspace, 2016-10-24:
I want to control crosshair with couple of potentiometers. X, Y dimensions of screen are -180,+180. ADC gives me 0..1023 range. It will be very handy if there were "map" function, same as in Arduino C.

bromagosa, 2016-10-24:
It is actually a good and simple arithmetic exercise to implement this one yourself:

[image]

[image]

Here's the block in case you want to import it into your projects: 
[map.zip](https://github.com/bromagosa/Snap4Arduino/files/547725/map.zip).

By the way, `map` is not the best name for this block, as in languages with higher order functions like Snap<i>!</i>, `map` usually refers to [something else](https://en.wikipedia.org/wiki/Map_(higher-order_function)).

closed 2016-10-24 by bromagosa

microspace, 2016-10-24:
Thank you. Is there a way to make this function avaliable at startup?

bromagosa, 2016-10-25:
Well, not at the moment... but it's just a drag and drop away :)