bromagosa/Snap4Arduino #94: Blinking goes out of sync

microspace, 2016-10-24:
I've attached 3 led to 9, 11 and 13 pin. 
After several minutes of synchronuous blinking, the 11th pin led start to blink out of sync with others.
Is it firmata issue?

bromagosa, 2016-10-25:
Hi @microspace.

I'll set up the same test today and let it run to see if it fails on me. This is actually a test I used to leave running overnight to confirm things worked as expected and it used to work fine.

Can you tell me more about your setup? What Snap4Arduino flavor are you using (GNU/Linux [32/64], Win [32/64], ChromeOS, Web?), and what kind of board?

Thanks! :)

microspace, 2016-10-25:
Original Arduino UNO, Windows 10 64.
But...
I can't reproduce to bug now.
I'm really sorry

closed 2016-10-25 by microspace