bromagosa/Snap4Arduino #96: Error while save project

aguscout, 2016-10-31:
snap4arduino present problem save project, no present action button browser , my version is 1.2

jguille2, 2016-10-31:
Which is your OS ?
I'm testing in Linux and Save function is ok
[image]

aguscout, 2016-10-31:
my SO is Windows, when press button the Browser no open   tree directory

bromagosa, 2016-10-31:
Please the page 14 of the Snap<i>!</i> manual about how to save projects: https://snap.berkeley.edu/SnapManual.pdf

Browser doesn't refer to "File Browser", but to an internal storage system for Snap. Just give the project a name and click Save, and it'll stay there.

Cheers,
Bernat.

closed 2016-10-31 by bromagosa

aguscout, 2016-10-31:
do you plan  saved projects to disk  in next versions or not ?

bromagosa, 2016-10-31:
Please check the manual, you can export projects to disk.

aguscout, 2016-10-31:
ok,  thanks