bromagosa/Snap4Arduino #97: Ultrasound error when open project import from http://blog.s4a.cat/2015/03/31/Ultrasound-Library-for-Snap4Arduino.html

aguscout, 2016-11-02:
hello, my error is when import xml 
my version snap4arduino is 1.2 on windows 10

This project has been created by a diferente app: 
                          Snap!
and may be incompatible or fail to load here.

you hava a version update  of ultrasonic library

bromagosa, 2016-11-02:
That's right, it needs updating.

Although the project does load and work normally, right?

aguscout, 2016-11-02:
load, but  the sensor dont receive señal and used with various HC-SR04

bromagosa, 2016-11-02:
Are you using the modified Firmata for ultrasound sensors? What Snap4Arduino flavor are you using? MacOSX, Linux, web, ChromeOS?

aguscout, 2016-11-02:
yes , i use firmata downloaded http://vps34736.ovh.net/Ultrasound-Snap4Arduino.zip

seteahconnoc, 2017-02-11:
Hello. I have got the same problem. Snap4Arduino just sends me a "0" from the block "ultrasound reading at '8'". I would be glad, if you could help me. I am using the same firmata that "aguscout" mentioned and:

Debian 3.16.39-1 (2016-12-30) x86_64 GNU/Linux
Snap4Arduino_desktop-gnu-64_1.2.2

I also checked out
Debian 3.16.39-1 (2016-12-30) x86_64 GNU/Linux
with the experimental web based version for the chrome browser
with no success.

Thank you, in advance.

jguille2, 2017-02-11:
Hi @seteahconnoc and @aguscout ,

I'm not using your related firmata, but I tested before and it works. I try to write down some commens and we'll see...

  - This firmata uses only one pin to echo/trig. Is it your case? Have you a 3pin HC-SR04? or has it 4 pins?
  - You can use [this other firmata](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md). With this you can use 3 or 4 pinned HC-SR04 (If you have 3, write the same number (pin) for echo and trig). You can also change (if the result is not good) the pulseOut configuration (by default it uses a 2-5µs puse and it's right).
  - Experimental web-based version does not work with these extra reporters!

Send your feedback and continue into this issue...

Joan

seteahconnoc, 2017-02-12:
Thank you very much, @jguille2. It 's working just fine.

---

My next step would be to use a Raspberry Pi instead of a notebook. I am a teacher and I am planning to use this in an educational project. Sadly (as you said), I cannot use the experimental web-based chrome browser version ("Inside: TypeError Cannot set property '202' of undefined"). That would be quit great. Otherwise, I have to try using the way, that @bromagosa describes here:

https://github.com/bromagosa/Snap4Arduino/wiki/Autonomy-via-CLI

I am not quite sure, if this is simple enough for my students, as @bromagosa is pointing out "you will need to have basic GNU/Linux skills to install and use this version."

jguille2, 2017-02-12:
Nice! I think we can close this issue.

I want to spend time in Web-based version and improve these expand functions... but there are so many things to do!!  Coming soon...

About your projects... I think the best way to work (to learn) with students is the Desktop version. They can program dynamically, debug, test... If you want the Raspberry Pi only for wireless connection, you can use a [bluetooth connection](https://github.com/bromagosa/Snap4Arduino/wiki/Autonomy-via-Blueetooth). Yes I know that you still need a PC. I hope in a few months we will have the android version...

Joan

seteahconnoc, 2017-02-12:
I also think, that we can close this issue.

You are getting my project quite right. Thank you for your efforts in improving Snap4Arduino. A web-based solution would be nice for multiple reasons. But I can understand you well enough.

bromagosa, 2017-02-14:
There is an inherent limitation imposed by Google Chrome in their extensions system that prevents the web version from running arbitrary JavaScript after load time. This prevents us and will always prevent us from using modified Firmatas in the web version :(

I can't wait for the day when WebUSB is mature and usable enough so that we can ditch every single version of Snap4Arduino and move it all to the web, but years need to pass before that's even possible.

closed 2017-02-14 by bromagosa