bromagosa/Snap4Arduino #99: adding new devices

najiJazzar, 2016-11-03:
As I was following this link  [http://blog.s4a.cat/2015/03/13/Extending-Firmata-for-Snap4Arduino.html](url).

I tried to add my own ultrasonic sensor block, which has two pins (trig,echo)unlike the one listed in the website : [http://blog.s4a.cat/2015/03/31/Ultrasound-Library-for-Snap4Arduino.html](url) which only have 1 pin.

I found that there are some javascript functions in the one you created (shown on pictures attached) that was not illustrated.
So please if you could give some hints or few steps to do in order to understand the logic of the javascript code. just some tips and I'll spend the time to read and understand.

[image]
[image]

jguille2, 2016-11-03:
Hi,
If you want to use that project, you must also change arduino sketch.

You can use [this other](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md), that have an Ultrasonic block with two pins (you can also use this with only one pin (the same in two slots)).

To understand the logic of this, you must look at block code (javascript) and arduino code. We use the SYSEX commands in the Arduino sketch to define new functions.

Joan

najiJazzar, 2016-11-06:
I've uploaded SA5Firmata2.ino to my Arduino mega, and tried the ping reporter for HC-SR04 ultrasonic sensor and the result was as shown in the picture.
_Inside:Error:
Please upload SA5Firmata2.ino a higher version_
Any ideas !!

[image]

closed 2017-01-11 by bromagosa