bromagosa/Snap4Arduino #101: Translation to Galician. Traducción a galego

tecnoloxia, 2016-11-08:
Quería aportar la traducción a galego. ¿Que ficheros debo traducir? Gracias

I wanted to contribute the translation to Galician. Which files should translate? Thank you

bromagosa, 2016-11-08:
Ola, obrigado! :)

Para la traducción de Snap!, deberías usar [esta plantilla](https://raw.githubusercontent.com/jmoenig/Snap--Build-Your-Own-Blocks/master/lang-de.js), mientras que para las cadenas de texto extras de Snap4Arduino, deberías usar [esta otra](https://github.com/bromagosa/Snap4Arduino/blob/master/src/s4a/lang-template.js).

Un saludo!

closed 2016-11-08 by bromagosa

tecnoloxia, 2016-11-08:
Podría disponer de la primera plantilla que envías en un idioma que no sea el alemán? Si la tienes en castellano o portugués me valdría.
Gracias

jguille2, 2016-11-08:
Hola @tecnoloxia,

Bernat te ha pasado la de Snap! en alemán porque es la oficial. Así, aunque mires las otras para orientarte, mejor que escribas sobre la 'oficial' (por si algún traductor no la tiene actualizada o lo que sea).

Puedes encontrar todos los archivos de traducción de [Snap!](https://github.com/jmoenig/Snap--Build-Your-Own-Blocks). Así tambien podrás preveer diferencias por las fechas de actualización (la [oficial](https://raw.githubusercontent.com/jmoenig/Snap--Build-Your-Own-Blocks/master/lang-de.js) hace 11 días, [portugués](https://raw.githubusercontent.com/jmoenig/Snap--Build-Your-Own-Blocks/master/lang-pt.js) hace 9 días y [castellano](https://raw.githubusercontent.com/jmoenig/Snap--Build-Your-Own-Blocks/master/lang-es.js) hace 4 años!!).

Los de Snap4Arduino están todos actualizados. La [plantilla](https://raw.githubusercontent.com/bromagosa/Snap4Arduino/master/src/s4a/lang-template.js) ya te la ha pasado Bernat. Puedes ver los [otros](https://github.com/bromagosa/Snap4Arduino/tree/master/src/s4a) (y concretamente [portugués](https://raw.githubusercontent.com/bromagosa/Snap4Arduino/master/src/s4a/lang-pt.js) y [castellano](https://raw.githubusercontent.com/bromagosa/Snap4Arduino/master/src/s4a/lang-es.js)).

Cualquier duda nos preguntas (que no sea de gallego :) ) Atentamente,

Joan

MMSequeira, 2016-11-08:
Hola!

A tradução para português está mais completa que a oficial em alemão. :-)
Se a usares, ficarás com uma tradução mais completa também para galego.

Abraços,

Manuel

tecnoloxia, 2016-11-09:
Ola!
Gracias por las respuestas. Los ficheros traducidos a galego están [en esta dirección de drive](https://drive.google.com/file/d/0B6CqFl4gn5RCWEVVNTdQTlYyWWs/view?usp=sharing).
No sé dónde hay que subirlos, así que os los paso por aquí.

tecnoloxia, 2016-11-09:
Cuándo puede estar disponible el software en galego? Estoy trabajando en una guía de prácticas y me gustaría contar con el idioma actualizado para las imágenes capturadas.

bromagosa, 2016-11-09:
Obrigado!

Ya he enviado la traducción al repositorio de Snap<i>!</i>. Supongo que estará disponible en la próxima release, aunque no puedo estar seguro de cuando será esto.

En cualquier caso, si es muy urgente, podríamos construir una versión con el Galego integrado para que puedas realizar las capturas.

tecnoloxia, 2016-11-09:
Si no os da mucho trabajo os lo agradezco, pues toda la documentación que estoy elaborando está en galego y me gustaría que las capturas estuviesen en el mismo idioma.
Si no puede ser nada, usaré otro idioma y esperaré.

tecnoloxia, 2016-11-09:
Una duda. He enviado dos archivos. El primero vale también para Snap?

bromagosa, 2016-11-09:
Exacto, el primero lo he enviado a Snap<i>!</i>.

A ver si tengo un momento para hacerte la versión... en qué sistema operativo trabajas?

tecnoloxia, 2016-11-09:
En Linux. Tengo instalado Linux Mint 64bits.
Moitas grazas! :-)

bromagosa, 2016-11-10:
Ola! Tienes una versión de prueba aquí:

http://snap4arduino.org/downloads/test/Snap4Arduino_desktop-gnu-64_1.2.tar.gz

Por favor, confirma que todo funciona bien! :)

Por cierto, tu traducción ya está en el Snap<i>!</i> oficial.

tecnoloxia, 2016-11-10:
Muchas gracias
Acabo de probarlo. La traducción se ve, pero no soy capaz de conectar con la tarjeta arduino. Con la versión anterior conecto sin problema. 
Además debo ejecutar el programa a través de launcher.sh. A través del fichero Snap4Arduino no funciona.

bromagosa, 2016-11-10:
Hmmm...

[image]

Seguro que tienes Firmata bien instalado en la placa? A mi me funciona perfectamente.

tecnoloxia, 2016-11-10:
No era eso. Debía tener algún conflicto con todo lo que tenía abierto. Reinicié el equipo y ahora me funciona.
Perfecto. Muchas gracias!!!