bromagosa/Snap4Arduino #104: Export as Arduino Sketch fails

DocCleaner, 2016-12-03:
I started a new arduino translatable project in Snap4Arduino 1.2.1 win 64, created a very simple program and tried to export it. Snap4Arduino just gives me an error messagge "Cannot read property 'toString' of null".

The simple project is attached, so is a screen capture of the error message.

[ExportTest.zip](https://github.com/bromagosa/Snap4Arduino/files/628811/ExportTest.zip)

[image]

jguille2, 2016-12-04:
Hi @DocCleaner,

The problem is in the second **set digital pin _x_ to _y_** block. _y_ value must be a boolean, and in your script is empty (lika a string)

You must change your
[image]
for
[image]
And this is all.

Joan

closed 2016-12-05 by bromagosa

DocCleaner, 2016-12-05:
Hi @jguille2 ,

thank you for the quick answer!
I'm a little embarrassed, but this solves my problem.
I tested the original script with a transitor and LED connected and it switched the LED on and off, so I thought my setting was the FALSE setting. (I'm just starting with snap...)

Thanks again,
Sven