bromagosa/Snap4Arduino #108: Leap Motion on local version

fzsigmond, 2016-12-14:
We are trying to use Leap Motion to control a robotic arm on Snap4Arduino. Back in January I could make it work, but now, when I click on any block from the Leap Motion library I get the message below:

Inside: Type Error
Cannot read property ‘frame’ of undefined

We are using Arduino IDE 1.6.9 and Snap4Arduino 1.2.1 (Mac OSX). StandardFirmata is loaded and the board connects fine.
Everything works well on the browser version, we get LeapMotion and Arduino working. The issue only happens when we use the local version of Snap4Arduino.

jguille2, 2016-12-14:
Hi @fzsigmond,
I have not any experience with LeapMotion, but from code I see:
  - Your error "cannot read property 'frame' of undefined" is normal before executing the _Leap_ block
[image]
  - _Leap_ block creates _window.controler_ object. Then, all the other blocs use _frame_ like a property of this _window.controler_

So, be sure tu execute _Leap_ block first. If you already did this, the problem is in this execution.

fzsigmond, 2016-12-14:
Thank you for your prompt response! The issue is weird, because the exact same code that runs on the s4a browser version, does not run on the downloaded s4a version.By the error message, could that be that there is a missing piece of code that handles the "leap" block on the downloaded s4a? 
Best, 
Fabio

jguille2, 2016-12-15:
One question, are your PC connected to Internet?
_Leap_ block use an online library at "crossorigin.me/http://js.leapmotion.com/leap-0.6.4.min.js"
[image]
Test this _http reporter_. If this report nothing... this is the problem.

If you need work without Internet connection, you can download this _js_ and change _http reporter_ URL to "http://localhost/..."

Joan

fzsigmond, 2016-12-15:
Hi Joan,

My computer is connected to the internet.
I get this message when I tried the reporter:

Origin: header is required

Could you walk me through the steps in order to download the code and change the appropriated code?

Thank you,

Fabio

jguille2, 2016-12-15:
Oh, I see...
This '_header is required_' apears in Snap4Arduino_local (Nor web version and nor Snap! online/offline).

I have no time now... but tomorrow we'll see
Joan

fzsigmond, 2016-12-15:
Exactly! 
Thanks for your time, send me the instructions when possible.

Fabio

jguille2, 2016-12-16:
Hi fabio @fzsigmond !
Test this:
Edit _Leap_block and change this URL 'crossorigin.me/http://js.leapmotion.com/leap-0.6.4.min.js' for this other 'js.leapmotion.com/leap-0.6.4.min.js'

I think, this will run ok.
Note: For the other versions (online, snaps...) you will need the original (with the crossorigin.me proxy).

Joan

fzsigmond, 2016-12-17:
Thanks Joan, that worked!
Is there a way to adjust this also on the Leap Motion file that installs when you select the install library from within the Snap4Arduino local version?

Best,

Fabio

bromagosa, 2016-12-19:
Hi @fzsigmond!

Please send me the modified library and I'll make sure to include that one locally :)

fzsigmond, 2016-12-19:
Hi Bernat,

Please see the modified file on the  attachment .

Best,

Fabio

bromagosa, 2016-12-21:
Hi Fabio, I'm afraid you can't send attachments to Github issues by email, you'll need to retry by using the Github UI.

fzsigmond, 2016-12-21:
[leap.xml.zip](https://github.com/bromagosa/Snap4Arduino/files/666428/leap.xml.zip)

bromagosa, 2016-12-21:
Thanks! It's already done, and will be available in 1.2.2, which I'll be publishing tomorrow :)

closed 2016-12-21 by bromagosa