bromagosa/Snap4Arduino #109: Makinh a INPUT_PULLUP block

hakanatas, 2016-12-15:
how can i create a block which is defined as INPUT_PULLUP ?

jguille2, 2016-12-16:
Hi @hakanatas ,
We could create JS blocks to use InputPullup, bet we need a newer firmata version for Snap4Arduino.
I've asked about firmata.js updates to @bromagosa. 
Soon, we'll comment when pullups will be supported.

bromagosa, 2017-01-11:
Sorry about this, I'll see if I can test Snap4Arduino with a more recent version of firmata.js this week.

Seven7au, 2017-02-07:
It would be great if you could enable the use of INPUT_PULLUP. I am currently developing some learning materials for students in junior school, but the inability to initialise the pinMode is a bit of an issue. Hope you can organise this soon. Best of luck.

bromagosa, 2017-02-07:
Actually, what used to be an issue for the vast majority of teachers was the obligation to initialise the `pinMode`, forcing them to explain to kids the need for this seemingly arbitrary command. I guess you can never please anyone ;)

What are you trying to achieve and why do you need to explicitly set `pinMode`s? Maybe we can find a solution for your case.

Seven7au, 2017-02-07:
I have designed a physical interface shield for the Arduino which has a button on it among other components. I did argue with the other developers that this button should have its own pullup resistor, however, the others sighted the internal pullup in the ATmega328. So you can see that not being able to use INPUT_PULLUP in this case means that the button on the shield is useless when used with the Snap4Arduino code platform. It is good not to have to initialise the pinMode, however, it is even better to have a choice. I hope there is a solution, because I selected Snap4Arduino over several of the other visual block platforms due to its comprehensive range of blocks and functionality. Thank you for your time with this issue.

jguille2, 2017-02-08:
Hi @Seven7au ,
I was waiting for the next Snap4Arduino version (with firmata.js updated) to see the better solution...

But if you need this feature now, we can work it out...

Test this block (digital PULLUP on x pin) and it will report the digital reading with a pullup configuration.

[pullup_block.xml.zip](https://github.com/bromagosa/Snap4Arduino/files/759488/pullup_block.xml.zip)

We'll be waiting your feedback... and then, we'll think about better implementations.

Joan

jguille2, 2017-02-08:
Excuse me @Seven7au ,

I see my block is not right without the last changes (on dev).

Don't test it. I'll see it later and I'll post the result.

Joan

hakanatas, 2017-02-08:
i will try it and that moment i see your mail. Like Seven7au , we are using
http://www.tinylab.cc/ in our labs and i cant use rotary encoder since
there is no pull up resistor in it. I also want to modify firmata to use 7
display segment and etc. But i think it will not be so easy (i am not a
programmer but trying to learn)

Seven7au, 2017-02-08:
Thanks Joan for your work on this, your efforts are very much appreciated.
Cheers, 
Seven

jguille2, 2017-02-08:
Hi @hakanatas and @Seven7au ,

I've made another block to implement pullup in our current version.
[pullup_block.xml.zip](https://github.com/bromagosa/Snap4Arduino/files/760636/pullup_block.xml.zip)
This is not a reporter. We use 'digital reading' block normally. But with this 'set pin to pullup/standard input' action block we can change the behaviour (pullup or standard input).
  - You can set the input config before or after the readings.
  - You can change this config dinamically.
  - If you use 'digitalWrite block', you turn to outputmode. After this, if turn to read (digitalRead) it will be in the standardmode (not pullup).

@hakanatas, In [Snap4Arduino.org](http://snap4arduino.org/), looking at 'Devices' you will find some info. You can use the related projects (or learn with them your way to do it). You can also ask for more features (I try to implement the most useful ones in [SA5Firmata](https://github.com/jguille2/SA5))

Joan

Seven7au, 2017-02-09:
Beautiful, works like a charm.
Thank you so much for your time spent on this Joan. You are a champ!

Do you if there are guidelines on the Snap4Arduino website regarding the use of screenshots of the platform in educational publications?
I am halfway through a guidebook which will be publicly available in late February/early March, and I want to make sure I am complying with the copyright wishes of the Snap4Arduino developers.

Best regards, 
Seven

bromagosa, 2017-02-09:
Hi Seven!

Snap4Arduino is a derivative of Snap<i>!</i>, and is thus also free software released under the AGPL, which is a CopyLeft license.

There are different points of view on whether screenshots are derivative works or not, but the practical answer is that nobody will ever come after you for using a screenshot of a free software product..

If Snap<i>!</i> was proprietary software, that'd be a completely different picture.

_tl;dr:_ No need to worry at all :)

bromagosa, 2017-02-09:
Actually, I've just found out there is no controversy at all when it comes to the GPL:

https://www.gnu.org/licenses/gpl-faq.html#WhatCaseIsOutputGPL

So yep, you can take a screenshot of any free software product and do whatever you want with it.

Seven7au, 2017-02-09:
Thanks for that.
One more question if I may:
I know how to create a 2D array - by placing lists within a list,
However how do a access elements within a 2D array, eg element[2][5] ?

bromagosa, 2017-02-09:
[image]

If you need to be doing this all the time, I'd recommend building your own block:

[image]

[image]

hakanatas, 2017-02-09:
Hi,
thanks you for pullup it works perfectly. I dont understand this javascript
code.

2 questions

1) how can i learn to write these kind of blocks

2) I have an lcd but connection is different in my Arduino set
www.tinylab.cc . So I cant use LCD or 7 display with sa5firmata. Do you
have any recommendation to start learning these kind of stuff. Or using
standard one is a better way :)

thank again

jguille2, 2017-02-09:
Hi @hakanatas,

1- Oh! I'm learning everyday a little more ;-) Really it's not a complex JS code... but we need to take care of all our environment: [Snap!](https://github.com/jmoenig/Snap--Build-Your-Own-Blocks) itself (stage, sprites... "this" context), [Snap4Arduino](https://github.com/bromagosa/Snap4Arduino) functions, [firmata.js](https://github.com/firmata/firmata.js) lib, [StandardFirmata](https://github.com/firmata/arduino) sketch and all our electronics and  [Arduino](http://www.arduino.org/) environment. !!
 You must look at all this repos: code, docs, issues... And everyday you learn something new :)

2- LCD -> Soon!!!
I was working (just yesterday) to integrate LCD (i2c and 6pins driver) into SA5Firmata. But (now) I want to check if I can manage this elements without any firmata modification (just sending basic commands from Snap!)
I'll post my results.

3- 7Displays
I have not any displays to test... If you post the specifications (which driver... or an arduino sketch example) I'll take a look.

Maybe better in other issue... [@bromagosa, we need a forum :pray: ]

Joan

bromagosa, 2017-02-09:
We have a prototype of a Snap<i>!</i> forum, and there's a Snap4Arduino subforum in there, but the Internet will probably be obsolete before we meet the requirements to have it go public... 

@jmoenig, @brianharvey, we need a forum.

Seven7au, 2017-02-09:
Thanks bromagosa,
Visual block code can be more complex than text based code in several situations.
I am gradually working out what I can do with all of the blocks.
Cheers, 
Seven

brianharvey, 2017-02-22:
@bromagosa  Sigh, am I the only one nervous about this?  I still think the social network stuff should happen all at once, but if I'm the only thing standing in the way, I guess we can do it.

bromagosa, 2017-02-22:
@brianharvey I think if we managed to get SSO working we could release the forums right away. I totally agree with you that a unified login is important, we don't have one in the Beetle Blocks social site and some users get (rightfully) confused. @cycomachead have you looked at this? I guess we'd need MioSoft to offer an SSO endpoint, is that right?

As for the rest of issues, I really think we can work them out after releasing, but that's just my opinion! :)

cycomachead, 2017-02-22:
We've now hijacked a thread :/, which I guess proves the need for a good forum.
We should try to do as few things as possible "all at once" - we don't have the capacity to handle many issues at a time...
In any case, for Discourse, all we need is the right endpoint and to agree upon some security keys. I think I've sent the emails to Jens / miosoft at least twice in the past couple years. :)
It's also (in theory) possible for us to host an endpoint which uses the MioSoft API and then acts as an intermediary. I haven't really looked into this significantly, but it should be possible to do - but I'm much rather not introduce too many points of failure.

bromagosa, 2017-02-22:
Got it, so would you rather wait for the new API to be finished or do you think we could do this right away?

cycomachead, 2017-02-22:
I think it makes to do it sooner-if MioSoft can easily support an SSO endpoint on the current API system, that would be the best. If not, I think we should just consider time frames for the new API and how long it might take to write an intermediate server.

Our challenges with the forum are primarily non-technical, (though the solutions to those problems will probably be). I think that when it comes to management and scaling a community it will be best to figure that out when we don't have to worry about other potential issues. 'Debugging' forum polices and cloud issues at the same time would be less desirable than doing them separately.

jguille2, 2017-10-27:
Ok,

We can close this issue.

Here we have many comments, but the issue (about pullup operation) is answered [here](https://github.com/bromagosa/Snap4Arduino/issues/109#issuecomment-278202467).

closed 2017-10-27 by jguille2