bromagosa/Snap4Arduino #111: <digital reading _> turning off digital pin?

Smtih, 2016-12-19:
When trying to read a digital pins state it turns that digital pin off? I assume this is not the intended behaviour?

jguille2, 2016-12-20:
Yes,
When I read a digital pin, this pin turns to Input mode. Then, if i connect no sensor, I will see 0V (low), because is not a Pull_up mode (is an standard input).
Is this your question?

Joan

Smtih, 2016-12-20:
I'm not sure. I have an LED connected to a pin (say pin 10)
If I do
[Digital Pin 10 <true>] (The LED turns on)
<digital reading (10) > (Gives back true, but the LED then turns off)

jguille2, 2016-12-20:
I try to explain...

You turn pin 10 to true (high). Led turns on. You do a digital reading on 10. Led turns off... so 10 is low (false). But you have a true reading! Yes, but only the first reading. If you take another reading, you'll see false (low).

It's the normal behaviour, because this processes are asychronous. We can say "your first reading of a true state, change this state itself to false", and so the reading is of the state before. I any case, first reading of asynchronous processes may be wrong. This is a  firmata behaviour. The first reading actives the reporting and so, its first reading maybe wrong (only the first!!!)

Joan

closed 2017-01-11 by bromagosa