bromagosa/Snap4Arduino #112: Firs time Using Snap4A

mrlsm40, 2017-01-03:
My greetings. 
My name is Márcio Rocha Lima and I live in Brazil. 
I´m building a robot (a little car) to use in a project of Master Degree in Math using Arduin Platform and S4A too. 
The software are ready to use Arduin Board directly (at least was what I understood), 
but I´m using Arduino Sensor Shield 5.0 in a Arduin Board. This situation brought a problem: some doors (mainly of motors) aren´t available to use. 
It was indicated to use Snap4Arduino to solve this situation. I will use a step motor too.
Is it possible use Arduin Board, Arduino Sensor Shield 5.0 and Snap4A together? Is it limited the number of devices that I can use in this software?
Thanks. 
Márcio.

bromagosa, 2017-01-03:
Hi Márcio. You can indeed use all motor outputs in any Arduino board with Snap4Arduino, pins are automatically configurable and you can use them all indistinctly.

Hope it works out for you! :)

Cheers,
Bernat.

closed 2017-01-03 by bromagosa

mrlsm40, 2017-01-03:
My greetings Bernat Romagosa
Thanks for answers my question.
But I have a new one...
Is it necessary download any library to use all possible devices or the program is ready to use after install in PC?
Thanks again.
Márcio Rocha Lima

jguille2, 2017-01-03:
Duplicate in open (yet) #113. We continue there...