bromagosa/Snap4Arduino #113: A new question

mrlsm40, 2017-01-03:
My greetings Bernat Romagosa
Thanks for answers my question.
But I have a new one...
Is it necessary download any library to use all possible devices or the program is ready to use after install in PC?
Thanks again.
Márcio Rocha Lima

jguille2, 2017-01-03:
Hi Márcio,
You only have to follow the installation instructions in [Snap4Arduino.org](http://snap4arduino.org/) (upload StandardFirmata sketch on your board, and download and run Snap4Arduino on your desktop).
Then, you'll have the basic actions to manage the board: read digital/analog inputs, write digital outputs (high/low), write pwm outputs (0-254) and use servos (0-180º servos or continous-rotation servos).
This means that you can use any motor connected to a driver. The driver configuration will be the rule you must follow. Some pins will manage the direction of rotation (digital outputs) and others will manage engine power (digital outputs wit pwm ~).

Ask us if you need more complex behaviour.

Joan

mrlsm40, 2017-01-03:
Hi Joan
Thanks for help me.
Best regards.
Márcio