bromagosa/Snap4Arduino #114: Using bluetooth system

mrlsm40, 2017-01-03:
My greetings.
Is it possible use a bluetooth system is Snap4arduino??
I need to use my robot in an arena but programming it (of course) in a PC.
Thanks again.
Márcio Rocha Lima

jguille2, 2017-01-03:
Yes!
BT module connected to pins 0 and 1 (the pins used by the usb serial connection), and configurated to work at 57600 bauds.
  - Pair your system (PC) with this BT
  - In linux, make a serial connection with this BT connection
  - In Snap4Arduino, click the connection button and choose the port of the BT serial connection (in linux, something like /dev/rfcomm0; In windows another comx)

This [instrucctions](http://ateneu.xtec.cat/wikiform/wikiexport/materials/crea3/hw/btooth#connexio_bluetooth_i_snap4arduino) (under construction and in catalan) will help you.

Joan

mrlsm40, 2017-01-03:
Thanks again Joan.

closed 2017-01-04 by bromagosa