bromagosa/Snap4Arduino #115: Print to OLED Ascii instead text

adryannn, 2017-01-05:
I was trying to move from an LCD 16x2 display to an Oled128×64 i2c display. I have change the driver, library and coresponding commands for the new library. But I have keeped the same code(xml) in app.
Everythings works fine, except the fact that instead of text string, he printing the ASCII code. 
If I replace the argv[] with a text in the printing function from firmata, when I call the function from app (snap4arduino) he's printing the text.
So, why app is passing the ascii instead of string, when with the same code but the old lcd everything was fine.
Many thanks for your feedback in advance!

Br,
Adrian

adryannn, 2017-01-05:
http://blog.s4a.cat/2015/03/24/LCD-Library-for-Snap4Arduino.html

This is the sorce of firmata code and the library for the old lcd.

bromagosa, 2017-01-09:
I'm not sure I understood your issue. If I got it right, you successfully managed to print text in the OLED, is that so?

Different devices work in different ways, that's entirely up to the manufacturer and we can't build a universal library that handles them all. Probably the LCD library expects ASCII codes and translates them into their respective chars, whereas the OLED one accepts chars right away.

adryannn, 2017-01-09:
First of all, thank you very much for your reply, and for everything that you have done for thr community. This have a very big impact in our curent/future education.

I was afraid of a such answer but more probably your answer is correct. As per current lcd snap4arduino library code, the text variable is splitted in chars and passed to firmata. In the firmata the char's are composed and passed to lcd print function. Which is lcd.print("text var") for both lcd and oled product.
In the end it is about passing a text variable from snap4arduino to firmata, so is not 100% clear for me why you cannot pass the entire text without to be splitted in chars.is this a limitation of communicatiom between snap4arduino and firmata?
 Once you have the text, you can do whatever you want in firmata with the existing lcd/oled library code.

Sorry for the long message, but hope now is more clear.

adryannn, 2017-01-09:
For clarification. When I said text variable, I'm refering to the String variable that you mention in snap4arduino block and you want to be printed on LCD.
In my opinion if you are able to share between snap4arduino and firmata any type of variables, you don't need to code inside the snap4arduino(java script blocks), you can code the function directly in firmata.

adryannn, 2017-01-09:
case 0x08: // Print
    for (int c = 0; c < argc; c++) {
       lcd.write(argv[c]);

So firmata is using the below function to share the string 

void sysexCallback(byte command, byte argc, byte *argv)

Which probably the argc is position of char in the string and argv is the array)

So basically the lcd is printing 1 char per time. But still why ascii when both lcd print method should received string.
Still confused. Any help will be appreciated.

jguille2, 2017-01-10:
Hi!
So many things to comment... :)

  - Codification is complex. LCD library accepts different types (strings, dec, hex...) but firmata protocol sends data in serial bytes chains. So we need Hexadecimal codification! Detail: we write an string into the block slot -> block code transform text into an array of characters (with HEX codification) -> This array is sent to the board -> Firmata.ino use this with the lcd lib.

  - In sysexCallback argc is the number of data bytes, and argv is the array of bytes.

  - With each lcd command, a string (not one character) is writen.

  - We can't compare those two sensors. One uses  a driver (with the lcd lib) and the other uses i2c protocol (only two arduino pins). Totally different!! To use Oled i2c we don't need any library...standardFirmata is ok... and you must use i2c commands (we can write this in a JS block).

Continue...

adryannn, 2017-01-10:
Thank you so much for the explanation. I'm not sure if you are refering to the lcd library that you have to import in firmata so that firmata can work with the lcd, or you are refering to the js block from snap4arduino.

I'm confused because the old lcd had also i2c adaptor (4 pins).

As I sad before, the oled is communicating well with firmata, because if I replace argv[c] in the lcd print function with a simple text "Hello" when the case 0x08 is called from snap4arduino, the oled is printing "Hello"

So my confusion is related to how this parameter argv[c] looks when the string "Yellow" is passed from s4a.
In my mind is argc have 6 char so it is an array argv[5]
Then until int c =6 is printing the full array but char by char since the print function is called for every level of pointer of array.

jguille2, 2017-01-10:
Hi,
  - lcd blocks and firmata_lcd go together. This firmata implements case 0x08 (that uses lcd lib) and the JS block call this case.
  - Maybe your lcd runs in i2c... but if you use this lib, you are not using i2c connection.
  - You use "Hello" string into firmata.ino. Then you are writing arduino code, using lcd lib. I said lcd command (print) accepts strings, dec, hex... But this is the final action! To communicate with this code, using firmata, we need to send encoded strings.
  - Certainly, in the end it writes characters (by the way, one after one). But our functions (lcd.print or js block) send strings.
  - If you watch at JS block code, when you send "Yellow" there will be a serial communications, and it'll send "Yellow" sending each character coded (charCodeAt()). Then argc=6 and argv is an array of 6 bytes.

But the problem? If you want to use LCD, I think the code is right and you have it. And if you want to use i2c connection (with lcd or with oled) I think this method has nothing to do.

Continue...
Joan
  -

adryannn, 2017-01-10:
Ok, so I think the confusion come from my side. 
In firmata, I have replace the lcd library, with a library specific to this oled. Still, I have keeped the 0x08 case, and just replace the lcd.print function, which the same as structure, but dif spelling (like a oled.print ("") instead of lcd.print("").
So maybe I need to convert argv[c] to a string var, and then pass that var to the lcd print.

bromagosa, 2017-01-10:
Exactly, that should definitely work :)

closed 2017-01-10 by bromagosa

adryannn, 2017-01-10:
Thank you for the confirmation! Hope I will manage to do that.

This is my last component that I have to add to the robot. And then just to prepare some templates of blocks in s4a.
After that a full open source project (an arduino robot made from lego and controled by snap4arduino via bluetooth) will be available for community. :)
It can be made by anyone for around 40 eur. Is adaptable (due to the lego tehnic parts), and contain sensors, (ultrasonic, line follower sens), speaker, led, led matrix, display, motors and other can be added.
Hope the community will enjoy it. And will be useful to learn electronics and programming.