bromagosa/Snap4Arduino #128: Not working on VPS

aytac, 2017-02-06:
Hi there,

I have been trying to run snap4arduino on our own VPS server. However, it gives as attached error.

- I don't use both experimental URL and mine one at the same time
- I have install chrome crx extension as required.
Basically, I have clone the snap4arduino repository and upload them in /s4a folder

http://opentoydesign.com/s4a/ here is the link. All other functions fully working but Arduino blocks.

Is there any way to sort this issue out?

Thanks
[image]

bromagosa, 2017-02-07:
Hi @aytac!

The issue most probably comes from the fact that yours is not a whitelisted domain. The plugin whitelists a bunch of domains, and opentoydesign.com is not among them. This needs to be like this if we want to make sure our plugin isn't the source of a security hole.

May I ask why do you need to host it in another domain? Maybe I can help you find an alternative, or add what you find missing to the main trunk.

aytac, 2017-02-07:
Hi @bromagosa,

Thanks for the quick response.
What we are trying to do is, we are going to create an [open source platform](https://opentoydesign.com) where users can create their own models/toys etc with our standard parts as well as they can [contribute to improve](https://github.com/opentoydesign) the library and models. Our main target will be Turkey and that is why we would like to use in our own server where we can change the default language and do translation for Arduino blocks.

Also users can run their design/models at the same platform which we currently try Scratchx web based experimental, scratch for web etc. However, Snap4arduino is the most reliable platform that works with Arduino on the web for us. 

If we can do these on online on Snap4Arduino, there would not be an issue on our side.

We would be happy to contribute on Snap4Arduino as well.

Thanks

Aytac

bromagosa, 2017-02-08:
I see. So if you contributed the Turkish translation to Snap4Arduino and I added a flag to define the language from the URL (`&lang=tr`, for instance), would that be enough for you? :)

bromagosa, 2017-03-21:
`--lang` has been added at 1afd0de.

Since there's been no answer to [comment 278266689](#issuecomment-278266689), and I'm quite sure this would solve @aytac's needs, I'm closing this.

Please let me know if this isn't enough.

closed 2017-03-21 by bromagosa