bromagosa/Snap4Arduino #129: tone()

Seven7au, 2017-02-10:
I know how to use the sound blocks to generate notes via the computer's sound generator, however, I have searched for an answer to how to use Arduino tone(), but have not found an adequate answer.
If I want to use an external buzzer to generate sound, is there a way that I can call the tone function?

jguille2, 2017-02-11:
Hi @Seven7au !

You can use [SA5Firmata](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md).

With [SA5Firmata2.0](https://github.com/jguille2/SA5/files/695087/SA5Firmata_2.0.zip) you can use a "pseudo tone()" function. It's made to keep using IR_library (using the same Arduino timer than tone()). The result is a blocker tone function, right to play "beeps" and more... but with some limitations.

You also can use [SA5Firmata_tone2.0](https://github.com/jguille2/SA5/files/695088/SA5Firmata_tone_2.0.zip). It's the same but this have not the IR library and then, it use tone() standard Arduino function.

If you test both versions, you can use the same blocks (the tone block is the same) and you can change Arduino sketch if you need to improve tone limitations. The idea of SA5Firmata is implement all the features in one sketch and then, Snap4Arduino user does not need to change arduino sketch at any time.

Joan

Seven7au, 2017-02-11:
Thanks for the information Joan, you have been most helpful.
I will test out both of these scenarios.
Cheers, 
Seven

Seven7au, 2017-02-11:
This may be a dumb question, but how do you add extra IF statements to an IF/ELSE block? I know how this is done in scratch etc. but can't seem to find a solution with Snap4Arduino.

jguille2, 2017-02-11:
Hi @Seven7au ,

The simple way is to nest another if-else block into the parent "else".
[image]

But Snap! is great and you can implement what you want, making your own blocks.

For example, if you go to Libraries and import "Multi branched conditional (switch)" you'll get and extra control blocks working like switch/cases syntax.
[image]

Another example, the library "Infinite precision..." (is only in Snap! but you can export and import into your Snap4Arduino) has a "short format if-else":
[image]

And the best! You can edit all this imported blocks. Then you can see their construction and change it (adapting to your needs)... and you will learn how to build your own.

Joan

Seven7au, 2017-02-12:
Thanks so much again Joan, you are a great teacher.
All the best, 
Seven

closed 2017-02-13 by bromagosa