bromagosa/Snap4Arduino #131: Using bluetooth 4.0

adryannn, 2017-02-16:
Hello,

Recently I have tried to connect a Romeo ble mini board with Snap4arduino. I have managed to change the uart baud rate to 57600 and to connect the board via onboard bluetooth to my pc.
But when I tried to connect it in snap4arduino it said the acces to port no. is denied.
I just want to check if there are any restriction for bluetooth 4 with snap4arduine, and it is not supported. In case not, then means the issue is on the config side.

Ps: via usb the board is connected snap4arduino, and everything works fine.

Many thanks for your help!

Best regars,
Adrian

bromagosa, 2017-02-20:
Hi Adrian!

I'm afraid I can't help you much there... I bought a BLE module myself to test this out, and couldn't manage either. I'd need to spend more time into it.

I'll be keeping this issue open so that I remember to work on it in the (near) future.

Sorry about that. In the meantime you could use an HC-06 module and follow [this tutorial](https://github.com/bromagosa/Snap4Arduino/wiki/Autonomy-via-Blueetooth) to get it working.

adryannn, 2017-02-20:
Hello.
Many thanks for your feedback. I have tested with several laptops/bluetooth adapters, and I have observed different behaviours. Finally I have found  something on the internet, and the problem looks to be due to the compatibility of the different adaptors. It seems that is working well with the smartphones/tablets, but if you want to connect it  with a PC, you need their usb bluetooth adapter 》 Bluno Link.
I will buy on of this, which is around 10$, and let you know if is workings.
I have also find the following comments on their site: just make sure your Romeo is peripheral and USB BLE Link is central mode. They will setup comm automatically.
You can find more useful details on the comment section for this adapter,on their site.

Hope this help!

Best regars,
Adrian

adryannn, 2017-02-21:
Ok, so I have managed to test it with USB BLE-Link adaptor.
Everything works fine. Just plug in the adapter, will be installed as an arduino uno and you will see it in the Device Manager-> Ports
Just power on the romeo mini ble board. The board will automatically connect to the bluetooth adapter (you will see the blue led lighthing) and then open the snap4arduino and connect it to the available com port.
In my case I previously change the baud rate of the romeo bord as per snap4arduino required (with AT commands from arduino ide -serial monitor).

If you need any other details, just let me know.

Best regards,
Adrian

bromagosa, 2017-02-21:
Glad you got it working! Another device working wirelessly with Snap4Arduino! :)

closed 2017-02-21 by bromagosa

aramperez, 2017-08-14:
Has anyone gotten this to work with an HM-10 BLE adapter (instead of an HC-06) and a new Windows machine that has BLE built in?

Thanks,
Aram

alanis480, 2019-08-12:
I also got the BLE-Link Bluetooth dongle to communicate with a Bluno board (Arduino Uno + BLE Bluetooth) and connect with Snap4Arduino. The BLE-Link is put into AT mode exactly like a Bluno (see https://wiki.dfrobot.com/Bluno_SKU_DFR0267#target_6) and must be set to Central (ie, Master) mode. Set baud rate to 57600, but communication for AT commands continues at 115200. Ensure Bluno is set to Peripheral (ie, Slave) mode and 57600 baud. Note that it is extremely difficult to see the blue LED on the BLE-Link, but it does come on automatically when Bluno is powered up.
Snap4Arduino now connects to BLE-Link and all starts to work. HOWEVER, it stops responding after setting a servo rotating, even though Snap4Arduino is still connected to Arduino. What's happening?