bromagosa/Snap4Arduino #133: help needed in creating desktop custom build

sudarshan395, 2017-02-19:
Hi @bromagosa @MMSequeira @hasso @jguille2 @pierre-rouanet @edutec  @awangenh I have a project, where I need a custom build of Snap4aArduino for desktop. I want to add my customs blocks in a new category in xml. Please help guys.

sudarshan395, 2017-02-20:
How can I have my own categories in smap4arduino and then add my own custom blocks

bromagosa, 2017-02-20:
You can build your own blocks and export them as an XML library as explained in the [Snap<i>!</i> reference manual](http://snap.berkeley.edu/SnapManual.pdf). To import them, you just need to drag and drop the file.

If what you need is to add categories, then you'll have to dig into the source code. Check out the objects.js module to see how the Arduino category has been added to Snap4Arduino.

Cheers,
Bernat.

closed 2017-02-20 by bromagosa

sudarshan395, 2017-02-24:
Thanks for replying @bromagosa . Can you tell me after creating all custom blocks , where i can save those xml files or where i can define them so that they can reside in my Application permanently and I would come up with my own custom Snap4Arduino desktop software

Sauri89, 2018-08-27:
@sudarshan395  can your resolve your question? i need to know can i import my xml when app opens, without i need import after open.

jguille2, 2018-08-28:
Hi @Sauri89 and @sudarshan395,
You can build a _desktop shortcut_ to launch Snap4Arduino with a _xml file_ loaded (it can be a library with custom blocks or a complete project).

You only need to inform the path: `--load=/path/to/your/file/file.xml`
Also, you can choose your language: `--lang=ca`

In linux, the _Snap4Arduino_ exec file is a desktop launcher, so you have to build your own desktop launcher using the _run_ file. Then, an example will be:
`./run --lang=ca --load=/home/userName/file.xml`

I think it is a good practice to build different _start points_ for Snap! and Snap4Arduino. This helps users a lot... and the sessions gain autonomy. You can hide blocks, create (or maybe only rename) others, add comments, maybe some example script ...

That's all,

Joan

Sauri89, 2018-08-28:
@jguille2  I need load the xml when built the app indeed, can i do that? for example
./build --platform=desktop/gnu/64 --lang=es --load=/home/userName/file.xml --run
or precharge with code before Snap4Arduino is launched?

Sauri89, 2018-08-28:
@bromagosa @jguille2  i read a lot of post with your solutions when is launching the app but

how i can get save my exe with my custom blocks in your solution because when i do

./run --load=home/sauri/Documentos/Desarrollos/Snap4Arduino/
src/Andromie/examples/MakerMiniBloques.xml

just for once time when i launched after these when i execute .desktop in linux the blocks are not appear 

i need something like that

./build platform=gnu/64 --load=path/file.xml --run ?

jguille2, 2018-08-28:
Hi @Sauri89 ,
Maybe is a typo in your post... but it must be `--load=/home/sauri...`. We need absolute paths.

I think you can pass the args to your _build_command using linux pipes. But I don't understand the idea.
To run _build_ you need a development environment (with git, nodejs...). Yes, you can build (test) hacks in the code, but do you need a shortcut to launch your tests? You can make a desktop (outside the git repo, maybe in your Desktop) to launch the `.../releases/gnu/64/run --load=....` Then you will build (only if you have made changes in the code) and after this, you will use the desktop launcher.

In anycase, I think we don't need more params in the build script.

Joan

Sauri89, 2018-08-28:
@jguille2  sorry my mistake i put 
./run --load=/home/sauri/Documentos/Desarrollos/Snap4Arduino/
src/Andromie/examples/MakerMiniBloques.xml

the main idea is only i have to do once time and when re open the app the blocks will stay there without did 
./run --load=/home/sauri/Documentos/Desarrollos/Snap4Arduino/
src/Andromie/examples/MakerMiniBloques.xml
 again, do you understand me? i well generate a build with my xml and pass another people but when they open it, the blocks will appear instead, if cant undertand me, maybe could be talk in spanish

jguille2, 2018-08-28:
Hi @Sauri89,

Sure, my Spanish is better :laughing: (and you, where are you from?) but we can continue in English keeping a public discussion.

- I think I understand now your problem. If you want to pass your Snap4Arduino (with the starting point inside), the absolute paths are a problem. But you have to do nothing with `build`... Use the `./run` launcher and a bit of linux scripting to do what you want to do.

  The basic idea is to use "$PWD" to write the current path.

  For example, you can add your _xml file_ to the Snap4Arduino folder (beside _run_ , _launcher.sh_ ...) and you can edit _launcher.sh_ adding to the last line your xml
`/"$BIN" --load="$PWD"/testing.xml`
  Then, your "normal " Snap4Arduino will be launched with your xml project/blocks.

- Only to consider... You can use Snap4Arduino online and you can share easily your projects, just sharing the url.

Is this right for you?

Ya dirás...

Joan

Sauri89, 2018-08-28:
@jguille2 

Ahh, Im from México, but i lived in Spain for 2 years, im currently in Merida, Yucatan, Mexico, well my linux skills are basic but for this proyect i had to use.

I thought something like that but not used PWD, thats my wrong... but i get another questions if you give me your e-mail, could be continue the disscussion in spanish, after that can comment in eglish here,, 

saludos.

jguille2, 2018-08-28:
Ok, so I guess this _launcher.sh_ hack is right for you.

If you want to email me, you can find my contact in [Snap4Arduino project page](http://snap4arduino.rocks/).

Joan

Sauri89, 2018-08-28:
@jguille2  

I send you an email, extracted from github because i cant get from snap4arduino page.

greetings