bromagosa/Snap4Arduino #135: Run a xml file from the command line

DavidGasku, 2017-02-20:
Hi, we're trying to integrate snap, and snap4arduino in our school. We would like to be able to automatically run the program in each computer of the computer's room, and load a lesson's xml as a starting point.

Is it possible to run the program with a xml file as a parameter? I've tried "./run ./lesson1.xml" and it doesn't work, but it seems it tries to load the file.

Any help?

bromagosa, 2017-02-21:
There's currently no way to do that, but let me check how hard it'd be to implement it and get back to you.

bromagosa, 2017-02-21:
It was super easy :)

It's fixed in ef55ba7233e487c09581aac6ff061c67100bd174, but I need to work a lot more on the dev version before releasing 1.2.3. How soon do you need this?

closed 2017-02-21 by bromagosa

DavidGasku, 2017-02-21:
Thank you, that was fast! Don't worry, we'll start electronics third term. I can even build it myself.
Btw, I find the http server very promising. I tested it with a little processing sketch, so I can fetch some variable fron the students and show it a screen. It opens lots of possibilities for my classes

bromagosa, 2017-02-21:
I never mentioned, the arg is `--load=path/to/your/project.xml`

Cheers!