bromagosa/Snap4Arduino #137: Allow using Arduino blocks while not connected

DavidGasku, 2017-02-24:
Hello, again. I'm trying a weird classroom setup, where all snap4arduino programs in student's computers connect to the teacher's computer, where the arduino is connected. In this computer, I have a ser2net daemon, with as many ports as students. This way, each student can't connect the arduino, and test his program. I've tested it and it is working quite well. Of course only one can be connected at any given time.

As we cannot afford one robot for each kid, we'll use one or two robots, so they can program them, and then connect to it. This will allow us to make projects with more expensive hardware in the future.

Sadly, you can't use arduino blocks when it is not connected. I guess the program need to establish communication first to know the arduino model. I would like to implement an option to allow programming for an arduino, which model can be given by a config file or so.

I know this is a very specific request, and I'm motivated enough to dive into the code to add this feature. I would like your opinion about its feasibility, and maybe the best approach to do this.

Long term, I think this would be a useful feature for many schools, so I would like to document it, and maybe ease the ser2net part with a GUI program.

Thank you very much! Snap4arduino has been a blow of fresh air for me, so many possiblities...

jguille2, 2017-02-24:
Hi @DavidGasku !

I think you don't need to change Snap4Arduino. You can make a base project to do simulation,

An example:

[image]

You build your own custtom blocks (the SIM blocks in the example). You will use only these blocks. These blocks make the simulation action and, if you select "board ON", also the real action (when you connected).

The best of this is not a generic example. You can adapt the idea into your projects. If you have leds, motors... the custom block maybe is a more concrete action (like "turn led --- to <>") and your simulation can be not numerical (lights in the scenario...)

If you want, take a look at [this old project](http://snap4arduino.org/run/#present:Username=jguille2&ProjectName=semafor). The idea of this is to show the work with kids (customizing blocks)... but it use also simulation. Child play with lights (virtual or real).

Joan

bromagosa, 2017-02-27:
Joan's solution is certainly a great option, does that fix your issue?

DavidGasku, 2017-02-27:
Yes, it seems a clever solution! I cannot open the project (cant conect to server, which, btw, is an recurring problem for me), but I have enough information to make the blocks I need.
I'll start with arduino in 3rd term (after easter holidays), it's just programming now. I'll share how it works.
Thank you very much.

bromagosa, 2017-02-27:
Cool, please do share how it goes with us :)

closed 2017-02-27 by bromagosa