bromagosa/Snap4Arduino #138: Adding own custom blocks in desktop build

sudarshan395, 2017-02-26:
Hi , I am writing again because i want to build my own custom blocks and save those permanently in arduino category so that i can have my own build of Snap4Arduino desktop application. Right now what i tried is adding my own block code in objects.js similarly like other blocks were added but after building , arduino category is disappeared . 
  Last time you told me to export blocks in XML library but i am unable to find your xml library where all the blocks are defined. I would like to know , if i want to add my own blocks permanently just like Phiro , what all files need to be edit

bromagosa, 2017-02-27:
The Arduino blocks are not in an XML, they're coded in `objects.js` and `threads.js`.

However, if you're in GNU/Linux, you could use the new (unreleased yet) `--load` parameter:

`./run --load myBlocks.xml`

Otherwise, loading an XML in Snap<i>!</i> is exactly one drag-and-drop away. Just drop your file into the Snap4Arduino UI and it'll load it.

Any of these help?

Cheers,
Bernat.

DavidGasku, 2017-02-27:
If you want several computers to load the same project, with blocks preloaded, the --load option works great. I succesfully started the program with a xml project in every computer in a classroom with a xml file. If this is what you want to achieve, the software we use is epoptes (for linux, although there are similar programs for other OS's), then you can launch the program with --load parameter. In my case, had to copy the xml first, as the load option didn't work for a network file path.

sudarshan395, 2017-02-27:
thanks @bromagosa  @DavidGasku  but i do not want something like what @DavidGasku  wanted . I want my own custom build for desktop and after digging into object.js , I have been able to add my new blocks in arduino category . Now the problem is , those blocks are not working . I am getting an error "cannot read property apply of undefined" whenever i click on my blocks .

jguille2, 2017-02-27:
Hi @sudarshan395 ,
I understand, you add your block into _objects.js_, and you see it. But, as @bromagosa said, you must also add your function definition to _threads.js_. 

`  Process.prototype.functionName = ... `

But I insist also that you don't have to hack Snap4Arduino code to obtain this result. It's not complicate to hack... but maintenance is another thing...
You can make your template (or templates) and save it in a xml project. Then you only have to open this template to begin your work. And with the --load option, your students only click in your icon and all will be done.

Joan

closed 2017-02-28 by sudarshan395