bromagosa/Snap4Arduino #139: Support Firmmata over TCP socket

olmerg, 2017-02-27:
I have the tested ESP8266 board, (wemos D1 mini)[https://www.wemos.cc/product/d1-mini.html] in serial mode with the snap4arduino. It work great, just after some ignorepin modifications in the firmmata firmware.

My question is about to support TCP to firmmata devices, which is do it in arduino is to change the stream object from serial to a TCP connection, which can be configured like server and client. There exist a way to do that in snap4arduino , add the support to this type of device (which expand to embeeded system and cell phone device...)

I suppose that it is necessary to modify (arduino.js)[https://github.com/bromagosa/Snap4Arduino/blob/master/src/s4a/arduino.js] but  [here](https://github.com/firmata/arduino/issues/257) is a example how to create this tcp stream in JS with jhonny-5.

bromagosa, 2017-02-28:
Hi! There's an experimental serial over TCP implementation in Snap4Arduino that was contributed by Ove Risberg. I have never managed to get it working myself, but I started writing a tutorial [here](https://github.com/bromagosa/Snap4Arduino/wiki/Autonomy-via-Wifi) based on Ove's explanations.

If you have experience in the field, maybe the tutorial will be useful to you. Please let me know if you manage, this is something I spent quite some time on and never got results :(

DenysZaytsev, 2017-07-11:
I was using ESP-LINK firmware for ESP8266 module. This works like a charm! Just don't forget to setup wifi module to use 57600baud rate!

xia01eng, 2018-01-14:
@DenysZaytsev Hey, friend! I'm really interested in esp-link. And I wanna put s4a and esp-link together. But I really do not know how to manage. Could you give some advice to get started?

DenysZaytsev, 2018-01-18:
@xia01eng Hi there. 
That's pretty easy, just flash esp-link firmware to your ESP and set baudrate to 56K. Then connect RX/TX and you'd be able to connect to your device via WiFi from Snap4Arduino.
In Snap4Arduino - select TCP connection, put there an IP address and port of your ESP (port should be 23).

aramperez, 2018-01-18:
Does Snap4Arduino support Bluetooth LE?

DenysZaytsev, 2018-01-18:
No. Only BT 2.0/2.1
Since BT LE (4.0/4.1) - uses completely different protocol - it doesn't
work as COM port.

aramperez, 2018-01-18:
I understand that BLE doesn't work as a COM port.

Would it be possible to add BLE support to Snap4Arduino?

TIA,
Aram

DenysZaytsev, 2018-01-18:
I'm not a developer of Snap4Arduino. Ask bromagosa :)

aramperez, 2018-01-18:
I'm hoping he will respond...

bromagosa, 2018-01-19:
Hi @aramperez and @DenysZaytsev,

I am currently only adding small improvements and bugfixes, and actually @jguille2 is doing most of the work in Snap4Arduino, as I'm very busy with other projects right now. Serial over BLE is [certainly](https://www.npmjs.com/package/ble-serialport) [possible](https://www.npmjs.com/package/ble-serial), but still not trivial to implement.

Let me see how hard it would be to add it, and whether I have any BLE-enabled Arduino boards laying around.

aramperez, 2018-01-19:
Hi @bromagosa,

Thanks for the information. I had not see the BLE Serial module, I'll take a look. I'm not sure where you physically are, but I might be able to send you a BLE board.

/Aram

DenysZaytsev, 2018-01-19:
No "BLE Serial module" exists. Serial over Bluetooth is just a BT profile.
BLE doesn't have those profiles. It is possible to create one, but no one
will do that for free))))

if you wan't Bluetooth profile with a serial connectivity you should use BT
2.1
Other way is to modify Snap4Arduino so if could communicate somehow with
BLE (but using some another protocol, not SERIAL).

Cheers,
D.

aramperez, 2018-01-19:
Hi @DenysZaytsev,

I believe you are trying to say that for BLE, there is no Serial profile (like there is for BT 2.x). But you can get a [BLE module](https://www.amazon.com/DSD-TECH-Bluetooth-iBeacon-Arduino/dp/B06WGZB2N4) that communicates with an Arduino (or other host) as a serial device.

The link in the response from @bromagosa creates a "virtual" serial port on the host computer. I'll be looking into the code.

Thanks,
Aram

DenysZaytsev, 2018-01-19:
Yes you're right - for BLE there is no Serial profile.

And NO, HM-10 won't communicate with Arduino (or other) as a serial device,
believe me, I tried it :)
Perhaps it is possible to use HM-10 to communicate as a serial device, but
I was unsuccessfull on that.

DenysZaytsev, 2018-01-19:
hmmm... acrually I've just googled a bit and found this article
https://evothings.com/control-an-led-using-hm-10-ble-module-an-arduino-and-a-mobile-app/
seems like might work out))
give it a try, just order on Aliexpress. You get the same modules with less
price. All those modules are Chineese)

2018-01-19 17:29 GMT+02:00 Deen <dennis.devil@gmail.com>:

> Yes you're right - for BLE there is no Serial profile.
>
> And NO, HM-10 won't communicate with Arduino (or other) as a serial
> device, believe me, I tried it :)
> Perhaps it is possible to use HM-10 to communicate as a serial device, but
> I was unsuccessfull on that.
>
>
>
>> Hi @DenysZaytsev <https://github.com/denyszaytsev>,
>>
>> I believe you are trying to say that for BLE, there is no Serial profile
>> (like there is for BT 2.x). But you can get a BLE module
>> <https://www.amazon.com/DSD-TECH-Bluetooth-iBeacon-Arduino/dp/B06WGZB2N4>
>> that communicates with an Arduino (or other host) as a serial device.
>>
>> The link in the response from @bromagosa <https://github.com/bromagosa>
>> creates a "virtual" serial port on the host computer. I'll be looking into
>> the code.
>>
>> Thanks,
>> Aram
>>
>>
>
>
>
> С уважением, Денис Александрович.
>
> No money - no honey!

DenysZaytsev, 2018-01-19:
if it would work out - you'd be able to use BLE with snap4arduino and
connect to your host via BT using your iPhone or whatever device you want.

2018-01-19 17:31 GMT+02:00 Deen <dennis.devil@gmail.com>:

> hmmm... acrually I've just googled a bit and found this article
> https://evothings.com/control-an-led-using-hm-10-ble-module-
> an-arduino-and-a-mobile-app/
> seems like might work out))
> give it a try, just order on Aliexpress. You get the same modules with
> less price. All those modules are Chineese)
>
>
> 2018-01-19 17:29 GMT+02:00 Deen <dennis.devil@gmail.com>:
>
>> Yes you're right - for BLE there is no Serial profile.
>>
>> And NO, HM-10 won't communicate with Arduino (or other) as a serial
>> device, believe me, I tried it :)
>> Perhaps it is possible to use HM-10 to communicate as a serial device,
>> but I was unsuccessfull on that.
>>
>>
>>
>>> Hi @DenysZaytsev <https://github.com/denyszaytsev>,
>>>
>>> I believe you are trying to say that for BLE, there is no Serial profile
>>> (like there is for BT 2.x). But you can get a BLE module
>>> <https://www.amazon.com/DSD-TECH-Bluetooth-iBeacon-Arduino/dp/B06WGZB2N4>
>>> that communicates with an Arduino (or other host) as a serial device.
>>>
>>> The link in the response from @bromagosa <https://github.com/bromagosa>
>>> creates a "virtual" serial port on the host computer. I'll be looking into
>>> the code.
>>>
>>> Thanks,
>>> Aram
>>>
>>>
>>
>>
>>
>> С уважением, Денис Александрович.
>>
>> No money - no honey!
>>
>
>
>
> С уважением, Денис Александрович.
>
> No money - no honey!

DenysZaytsev, 2018-01-19:
Actually I remember the reason why HM-10 didn't work out for me - When I
connected my laptop with HM-10 module (paired) - I didn't got any Serial
Port appeared in my Device Manager.
That's why I thought that it just doesn't support Serial Profile and
there's no way to communicate via Serial.

2018-01-19 17:33 GMT+02:00 Deen <dennis.devil@gmail.com>:

> if it would work out - you'd be able to use BLE with snap4arduino and
> connect to your host via BT using your iPhone or whatever device you want.
>
> 2018-01-19 17:31 GMT+02:00 Deen <dennis.devil@gmail.com>:
>
>> hmmm... acrually I've just googled a bit and found this article
>> https://evothings.com/control-an-led-using-hm-10-ble-module-
>> an-arduino-and-a-mobile-app/
>> seems like might work out))
>> give it a try, just order on Aliexpress. You get the same modules with
>> less price. All those modules are Chineese)
>>
>>
>> 2018-01-19 17:29 GMT+02:00 Deen <dennis.devil@gmail.com>:
>>
>>> Yes you're right - for BLE there is no Serial profile.
>>>
>>> And NO, HM-10 won't communicate with Arduino (or other) as a serial
>>> device, believe me, I tried it :)
>>> Perhaps it is possible to use HM-10 to communicate as a serial device,
>>> but I was unsuccessfull on that.
>>>
>>>
>>>
>>>> Hi @DenysZaytsev <https://github.com/denyszaytsev>,
>>>>
>>>> I believe you are trying to say that for BLE, there is no Serial
>>>> profile (like there is for BT 2.x). But you can get a BLE module
>>>> <https://www.amazon.com/DSD-TECH-Bluetooth-iBeacon-Arduino/dp/B06WGZB2N4>
>>>> that communicates with an Arduino (or other host) as a serial device.
>>>>
>>>> The link in the response from @bromagosa <https://github.com/bromagosa>
>>>> creates a "virtual" serial port on the host computer. I'll be looking into
>>>> the code.
>>>>
>>>> Thanks,
>>>> Aram
>>>>
>>>>
>>>
>>>
>>>
>>> С уважением, Денис Александрович.
>>>
>>> No money - no honey!
>>>
>>
>>
>>
>> С уважением, Денис Александрович.
>>
>> No money - no honey!
>>
>
>
>
> С уважением, Денис Александрович.
>
> No money - no honey!

aramperez, 2018-01-19:
I'm actually using an HM-10 clone called a JDY-08 I ordered via Aliexpress. A HW engineer and I designed a new board based on a Maple Mini with 2 H-Bridges and the JDY-08 for a low cost robot kit (the STEAMbot) we're bring to market soon. Here's a [video](http://www.thesteamtrain.cc/wp-content/uploads/2018/01/My-STEAMbot-480p.mov) showing a STEAMbot being controlled by an iPhone. The app, "My STEAMbot" is available now for iOS and Android.

DenysZaytsev, 2018-01-19:
Wow:) Pretty cool!

Actually me too!))) A hardware engineer I mean))
We decided not to design a new board for our product, but use wiring
instead (it's more cost effective for us and better choice for the product
as it's a self assembly construction kit for kids - more fun for kids to
connect couple of small boards together). You can take a look for more
details on our site
http://www.ecobot.com.ua/product/buggy-4x4

We were thinking to add iOS support, but as for now we're focused on new
models development. And now I believe we are going to move off BT+Arduino
and use WiFi ESP modules instead. They are much more powerful and cheaper.

DenysZaytsev, 2018-01-19:
Ahh sorry, forgot to tell that "ВІДЕО" stands for "Video" (our page still
doesn't have English)... or here is the direct Youtube link
https://www.youtube.com/watch?v=EIo6A-fF9m4

2018-01-19 17:55 GMT+02:00 Deen <dennis.devil@gmail.com>:

> Wow:) Pretty cool!
>
> Actually me too!))) A hardware engineer I mean))
> We decided not to design a new board for our product, but use wiring
> instead (it's more cost effective for us and better choice for the product
> as it's a self assembly construction kit for kids - more fun for kids to
> connect couple of small boards together). You can take a look for more
> details on our site
> http://www.ecobot.com.ua/product/buggy-4x4
>
> We were thinking to add iOS support, but as for now we're focused on new
> models development. And now I believe we are going to move off BT+Arduino
> and use WiFi ESP modules instead. They are much more powerful and cheaper.
>
>
>> I'm actually using an HM-10 clone called a JDY-08 I ordered via
>> Aliexpress. A HW engineer and I designed a new board based on a Maple Mini
>> with 2 H-Bridges and the JDY-08 for a low cost robot kit (the STEAMbot)
>> we're bring to market soon. Here's a video
>> <http://www.thesteamtrain.cc/wp-content/uploads/2018/01/My-STEAMbot-480p.mov>
>> showing a STEAMbot being controlled by an iPhone. The app, "My STEAMbot" is
>> available now for iOS and Android.
>>
>>
>
>
>
> С уважением, Денис Александрович.
>
> No money - no honey!

aramperez, 2018-01-19:
Thanks. I'm mainly a SW engineer but know enough about HW to be dangerous ;-)

I like your Buggy 4x4.

Reach me directly at aram.perez at thesteamtrain dot cc so not to clutter this issue on GitHub.

DenysZaytsev, 2018-01-19:
Sure, I'll send email.

jguille2, 2018-02-13:
Closing this issue...

Firmata over wifi by ESP8266 is running ok.

See [wiki docs](https://github.com/bromagosa/Snap4Arduino/wiki/Autonomy-via-Wifi) and [this related](https://github.com/bromagosa/Snap4Arduino/issues/224) issue for more details. Soon, more details into the wiki

closed 2018-02-13 by jguille2