bromagosa/Snap4Arduino #140: Troubles instaling Snap4Arduino Windows XP

jfonol, 2017-03-02:
We tried to instal Snap4Arduino v 1.2.2 in a Windows XP systems because still a lot of computers with this OS in catalan public schools. Some troubles appear when we try to run the program:
"No se encuentra el punto de entrada del procedimiento ReleaseSRWLockExclusive en la bliblioteca de vínculos dinámicos KERNEL32.dll"
Some add to instal this version or a previous one that works in Windows XP?
Thanks a lot.

jguille2, 2017-03-02:
Hola @jfonol,

    Mirem de donar resposta de seguida. Tenim molt present la realitat de les escoles.

The problem is nwbuilder (and more other things) are not supporting XP. And then we can't offer this support to Snap4Arduino current versions.
We wanted to have published a final XP compatible version... but we wanted to add to it more updates... and we didn't find the time...

I think last XP compatible version is 1.1.6-beta-RC2 (you cant download it [here](http://vps34736.ovh.net/snap4arduino/19-08-2016/Snap4Arduino-win32.zip)). I'll test this later, and I'll try to find time to make a better version for XP.

Another option, the opportunity to migrate these old machines to Linux. We have Linkat! And we have a perfect [Robotic Linkat distro](http://linkat.xtec.cat/portal/index.php?module=news&func=display&sid=182)!! It is a light distro (ok for old machines) and is ready to snap! (snap4arduino installed and permisions added to run with arduino boards directly).

Salut!
Joan

jfonol, 2017-03-02:
Gràcies per la resposta!
--
Joan Fonollosa Massana
Departament de tecnologia de l'Institut Montserrat

http://institutmontserrat.cat
jfonollo@xtec.cat

Copèrnic, 84
08006-Barcelona

jguille2, 2017-10-29:
Hi @jfonol,
I'm sorry to answer with so much delay :(

I want to find time to try to build an XP compatible version... but I haven't found it yet. I don't close this issue... and we will considere this later... but I don't assure we'll build it.

But,  apart our recommendation to use Linux for your old machines (and for all... :smile: ...  [I'm a Linux enthusiast]), I want to point you to our **Chromium version**. Desktop versions are better, but now (with the 1.2.4 release), we can work [online](http://snap4arduino.rocks/run/) and connect to Arduino with the [chrome-chromium](http://snap4arduino.rocks/#plugin) plugin in a good enough way. 

This is a workaround to use Snap4Arduino for XP users. Also, Chromebook users are using this version.

Joan

JorgeGlade, 2018-03-21:
Gracias Jguille, problema solucionado con la versión 1.1.6-beta-RC2

jguille2, 2018-03-21:
Ok,
I close this issue.

I only want to insist that now, _online_version with _chromium plugin_ is a better option to connect to the board.
It is a pity to miss all the improvements that Snap has had!
Joan

closed 2018-03-21 by jguille2

JorgeGlade, 2018-03-23:
Thank you but my conection is very very slow!!

jguille2, 2018-03-24:
Hola Jorge @JorgeGlade,

Are you using _chromium plugin_ or the old 1.1.6 version?

I suggest you to use _chromium plugin_ if you are still working in XP.

Continue...

JorgeGlade, 2019-08-15:
¡Hola!

Acabo de firmar la petición ".@SenadoArgentina ‏ : Aprueben el desafuero de
Cristina Kirchner" y me gustaría que te sumaras a ella.

Nuestro objetivo es conseguir 300,000 firmas y necesitamos más personas que
la apoyen. Podés tener más información sobre la misma y firmarla acá:

http://chng.it/BFJFDgLtrz

¡Muchas gracias!
jorge enrique