bromagosa/Snap4Arduino #141: Tone()

Seven7au, 2017-03-09:
Does anyone have a tone block which they have developed? To play a tone on an external buzzer. It would really help me out.

jguille2, 2017-03-09:
Hi @Seven7au ,

You can take a look at [this project](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md).

The problem is StandardFirmata has not ToneFunction implemented. Then you have to run block that use an extended firmata.

In this project, you can use SA5Firmata_tone 2.0, implementing Tone standard function into Arduino. The main version (SA5Firmata) use a blocking (more limitated) tone function that allows to use it with IR libraries (IR and Tone use the same timer).

Joan

Seven7au, 2017-03-10:
Wow, you have sure done a lot of work there. Well done, and thanks for your help again Joan.
Cheers, 
Seven

bromagosa, 2017-04-03:
Thanks @jguille2 for responding to this one too :)

Closing it now.

closed 2017-04-03 by bromagosa