bromagosa/Snap4Arduino #142: File permissions for cordova

sudarshan395, 2017-03-11:
Need help to add file permissions for cordova to import and export blocks . How to add file permissions to cordova ? We tried but we are not able to find the directory where we can add the plugin
We used this instruction 
cordova plugin add cordova-plugin-file

It says this is not a cordova based project . How to find that directory , we are only getting Snap4Arduino.apk.

bromagosa, 2017-03-15:
Yep, I was planning to work on this soon.

Please bear in mind that the Cordova version is not published yet because it's unfinished. Many things are missing and it's not recommended for anybody to use it yet.

The pre-alpha will hopefully come out by the end of April.

bromagosa, 2017-05-11:
This is implemented now :)

closed 2017-05-11 by bromagosa