bromagosa/Snap4Arduino #144: Can we import xml file containing custom blocks in Snap4Arduino on launch of the application?

rhlgupta, 2017-03-17:
Is it possible to get our blocks imported as soon as we launch the Snap4Arduino.exe?

Where we need to add import function and make these changes?

Even I tried to load the xml using your fix in ef55ba7, which you did in issue 135. But it is also not working.
I added path like:
 --load= /home/hadoop/Snap4Arduino/src/platforms/desktop/root/desktop/blocks.xml

jguille2, 2017-03-17:
Ops!

Assuming you are using a dev version (builded from github), I see there is problem in the launcher.

It runs ok if you have English as Snap! language. You can test this. Change to English and close Snap4Arduino. After this, if you run the langher (with --load=path\file.xml).

The problem is not easy (is not in the Snap4Arduino side) because is part of the Snap! behaviour (when language is different of English, snap reloads a new project to ensure the translation of all IDE elements (project name ('untitle'), sprite name ('sprite', 'objecte'...))

We will see what needs to be done.

Joan

rhlgupta, 2017-03-17:
So is there any solution to get my blocks available just after the launch of snap4arduino application?

or is there a place where I can add my blocks.xml code to get it in the application itself?

Urgent help needed

bromagosa, 2017-03-17:
`--load` is not issued at build time, it's issued at launch time:

```
./run --load
```

or, if you're in windows:

```
Snap4Arduino.exe --load
```

rhlgupta, 2017-03-17:
@jguille2 @bromagosa @MMSequeira @hasso @pierre-rouanet 

How can I import blocks at runtime or launch or Snap4Arduino application without asking my user to drag blocks in the interface or select the import file?

I just want that as soon as Snap4Arduino application launches it should have my blocks in it.

bromagosa, 2017-03-17:
Just create a shortcut that calls `./run --load=path/to/your/blocks.xml` or `Snap4Arduino.exe --load=path/to/your/blocks.xml`, depending on your operating system :)

bromagosa, 2017-03-21:
Since there's been no answer I'm assuming my last suggestion worked. Closing this now :)

closed 2017-03-21 by bromagosa

szjenoko, 2018-03-09:
Hi,
I tried it with Snap4Arduino v1.2.5 on Windows 10.
It works for me only if I try to load a project xml.
It doesn't work if I try to load an xml contains only blocks (a block library).
It could cause the confusion previously in this message stream.
Is it a bug? Do you have any suggestion how can I load a block library?
Best Regards,
Jeno

bromagosa, 2018-03-09:
Hmm... this used to work.

@jguille2 any idea what could have happened?

jguille2, 2018-03-10:
Oh, I'm sorry... I think this is my fault.

Adding _Jr_ launcher, I fixed some other issues of the _load launcher_ (using _--load_ and _--lang_ tags together).

But I thought that _--load_ was only for loading projects!!

So, I'll try to fix this tomorrow (it seems easy... but asynchronous launching is more complex if I combine two params).

@szjenoko , I'm sorry. In the meantime, you know the workaround is convert these files to projects (load libs into empty projects and then, save these projects).

Pending...

szjenoko, 2018-03-10:
No problem.
I use this workaround, but there are some disadvantages of this way.
My main problem is that students must use save as... function to rename the
empty project. If they forget it, may overwrite their previous projects.
Can I set a project property in the xml file to force naming project, as if
I create a new project. (I tried to name it as "Untitled", "", and tried to
remove the name property, but these ways don't work. I haven't checked the
code yet.)
Thanks your help in advance.

jguille2, 2018-03-10:
Hi @szjenoko ,
I suggest a _hack_ to workaround the problem of the "Save as" require.
- Open your xml (the project with your lib (custom-blocks)).
- Choose the name property as you want (maybe name="Untitle")
- Add this code between the `<scripts></scripts>` tags:

```javascript
<script x="60" y="42"><block s="receiveCondition"><l><bool>true</bool></l></block><block s="doRun"><block s="reportJSFunction"><list></list><l>ide.projectName = undefined;&#xD;this.scripts.children[0].destroy();</l></block><list></list></block></script>
```

That code deletes _projectName_ and, after this, deletes itself. Then you will have an empty project (without any script) and without a _projectName_ defined. So, normal "Save" will ask you for the project name.

I hope this is useful for you.
Joan

szjenoko, 2018-03-10:
What a great hack:)
It works perfectly and it's enough for me.
Thank you very much!

jguille2, 2018-03-12:
Hi @szjenoko 

Only note that this is fixed for next version (and also dev version from github).