bromagosa/Snap4Arduino #145: ultrasonic problem

basilhsk7, 2017-03-18:
hello,
i try to use ultrasonic with snap4arduino
(hc-sr04)
i did everything in instructions for ultrasoundfirmata etc
i import the xml woth ultrasonic block
it works,butttt
BIG BUT it has so much delay from reading to reading
for example i take the read 7 then it goes to 0 for 2-3 secs and then it displays the new reading
how i fix this delay and take readings constantly????
plz help me!!

jguille2, 2017-03-19:
Hi @basilhsk7 ,
I think you are using [this _edutec_ project](http://blog.s4a.cat/2015/03/31/Ultrasound-Library-for-Snap4Arduino.html).
You can contact this team about this problem...

but if you want, test [SA5Firmata](https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md) (it has the ultrasonic ping feature). I think it has not problems with delays. And any problem with it, you can report it.

Joan

basilhsk7, 2017-03-19:
hello @jguille2,
in the past hour i try to work with sa5,
but i didnt make it to work
I load all the libraries and i upload the code to arduino via IDE
everything seems ok but when i try to connect arduino in snap4arduino menu i take the msg
"could not talk to arduino in port COM3 check if firmata is loaded"
any ideas?

jguille2, 2017-03-19:
Hi @basilhsk7,

I do not have a hypothesis with the data that you comment.

I know many people are using SA5Firmata with no problem.

What can you do?
  - Sure, you have tested connecting and disconneting the board, or reseting it. We know sometimes, after the 'firmata missing' message (in any firmata) we have to close and open Snap4Arduino (or try to connect another sprite).
  - I guess you have checked COM3 is the right port.
  - Are you connecting by USB or by Bluetooth? If you connect by BT, I think Windows (I do nou use it) create two COM ports, and only one is 'connectable'.
  - Another test will be reload firmware, and check IDE messages. You must check IDE is using your custom (that are in your Arduino user folder) libaries (also the custom firmata loaded, to be sure is the same version).

I can not think of other problems. If it keeps ko, reload StandardFirmata to see if it is running ok.

Joan

basilhsk7, 2017-03-20:
Thank u for trying to help resolve my problem,
i already did all that u said above and nothing .
with standard firmata everything is ok
when i upload the custom snap4arduino wont recognise my board.
i even try it in a new pc without arduino IDE and intall it from begin with your custom libraries only and still nothing.
also i connect it with usb not with bt.
i will try the best to explain what i do step by step so maybe u will be able to help me.
i download the sa5 https://github.com/jguille2/SA5/blob/master/firmata/SA5Firmata.md
from the download section
i unzip it and copy the libraries in my arduino folder(i deleted the previous libraries)
i upload the sa5firmata2.ino into my board
after its uploaded i close the IDE and i open snap4arduino i go to arduino section and i press connect to arduino
and then i get the above message i wrote in the start..
i dont think im doing something wrong and i cant figure it out whats the problem
as i said i also did the same proccess in a new pc and i get the same error...!!!

jguille2, 2017-03-20:
One question (to try to test it)

What is the version of the Arduino IDE you are using?

basilhsk7, 2017-03-20:
version 1.8
and my board is arduino uno

jguille2, 2017-03-20:
I'm sorry but I have not any hypothesis.

I've tested with the last 1.8.1 version and all runs ok.

basilhsk7, 2017-03-20:
which OS u are using?

jguille2, 2017-03-20:
Linux (always :))

But is tested in others. We have a community of teachers using this, so is tested in different SO (also macs, w$...) and different machines.

basilhsk7, 2017-03-20:
status update,
i make it work
i did the same f@@@ process in linux and it just worked fine
FOR GOD SAKE
i do not know what was wrong 
2 windows machines and failed ,1 with clean install from 0 
and in linux just worked with the same download and same step by step process
i used the tested version of IDE 1.6.9
i really dont know what was wrong in windows machines
i cant thank u enough for your effort to help me
If i have any questions about sa5 where i can contact with u?

PS. the funny thing is after it worked in linux i just plugged in the windows machine i was trying to work at the begging and the sn4ar recognized it immediately
SO BIZARRE!!!!

jguille2, 2017-03-20:
Eureka!

Oh, Windows world is too complicate for me :) But, as I said, we have many windows users... and they haven't report this issue. (Mysteries of computing).

Of course, you can report anything to [SA5 project](https://github.com/jguille2/SA5). And if the problem is more generic, sure you'll find me into this Snap4Arduino 'forum'.

Joan

closed 2017-03-21 by bromagosa