bromagosa/Snap4Arduino #147: "analog/digital reading" watcher

maprzybylla, 2017-03-21:
Hi, is there any way to keep the “watchers” for input values after a program was closed and is opened again? What happens is that instead of e.g. “sprite analog reading 1” I only see “sprite 0” when opening a saved program later (screenshot attached). I can’t delete or reconnect it. Are you aware of this problem? Is there a known workaround?

Kind regards,
Mareen
[image]

bromagosa, 2017-03-21:
Hi Mareen, I had never noticed this until right now. I'll take a look, thanks for reporting it!

bromagosa, 2017-03-28:
It's fixed in the dev version now. I should definitely start thinking about releasing 1.2.3 soon.

commit e7d5b547 (2017-03-28): fixes #147
closed 2017-03-28 by bromagosa