bromagosa/Snap4Arduino #153: Send Snap4arduino code to Arduino and Unplug

hiquemonnerat, 2017-04-04:
Is it possible to send Snap4Arduino code to Arduino and have the code be written in Arduino´s memory so that we could unplug the board from the computer ...like the "Upload Sketch" from Arduino IDE?  That would be very helpful for mobile robotics applications.

jguille2, 2017-04-05:
Hi @hiquemonnerat ,
Two things about this:
#### Snap4Arduino feature to create Arduino translatable projects
  - In the "File menu" you have the option to create a _New Arduino translatable project_
[image]
  - Then you begin a new project with a reduced blocks palette.
  - You can connect to an Arduino and work in the Snap4Arduino way (using and programming dynamically).
  - But to build an Arduino sketch, you must begin the script with the _green flag_ and you'll have the option to _export as Arduino sketch..._
[image]

#### Orientation and Limitations
  - We built this feature as an extra tool. The idea was to give the ability to create basic sketchs from Snap (where we are really programming without any limitation).
  - The result is more or less good, using only the blocks offered by the new translatable project.
  - But this is not the idea of Snap4Arduino. And many things will not be possible!
  - Of course, we have more features than the board has. We have  screen, keyboard, mouse... We can paint, display things, connect to Internet... But maybe this is not the most important.
  - We program Snap in a different way. We have parallel and dynamic programming. We can send messages betwen  scripts and objects. We can use custom blocks. Nobody should expect all this to be translated into Arduino!!

Joan

bromagosa, 2017-04-05:
You may also wanna take a look at all the ways you can get rid of the cable, if that's what you're after:

https://github.com/bromagosa/Snap4Arduino/wiki/Getting-rid-of-the-cable

C doesn't have concurrent threads, lambda, first class lists, lists of lists, sprites, sounds, pens or messages, so it's virtually impossible to translate one to the other.

hiquemonnerat, 2017-04-10:
Hi @jguille2 
Thank you for letting me know about this, and also for writing about it´s limitations. 

Hi @bromagosa Thank you as well, I love the article you wrote. I will try some of the options after giving @jguille2 idea a shot. 

Question: It´s my first time posting here, should I close this issue? 

All the best, Henrique

jguille2, 2017-04-10:
Yes, it can be closed.

Really, we need a forum... but github _issues_ is our current workaround.

closed 2017-05-08 by bromagosa