bromagosa/Snap4Arduino #154: Bluetooth problem connection

IsaacFernandezUgaz, 2017-04-04:
I had a problem using bluetooth HC-06, sometimes snap4arduino connects ok with bluetooth and other times doesn't, the bluetooth and the arduino are connected to two litio batteries

(http://www.ebay.com/itm/Elfeland-18650-3-7V-3800mAh-Li-ion-Rechargeable-Battery-Charger-For-LED-Torch-/381973399137?hash=item58ef638661:g:3TwAAOSw32lYqsVc) 

so bluetooth is connected to 7,4V and 3800mAh through a sensor shield 

(http://www.ebay.com/itm/Arduino-UNO-MEGA-Duemilanove-Sensor-Shield-V5-Digital-Analog-Module-Servo-Motor-/262702939294?hash=item3d2a50bc9e:g:tVIAAOSwcUBYGuct) 

where there´s a jumper to use external voltage. When snap4arduino doesn't connect, appears a message saying that couldn't find arduino at com port or just keeps searching for a long long time, so I have to close and opend snap4arduino or testing bluetooth with an arduino code. 

I´m using Windows 10 64bits, I'll be very grateful if someone can help me. Thanks in advance.

jguille2, 2017-04-05:
Hola Isaac!!

Two comments:
  - When sn4a does not connect, try to 'reset' the board. Sometimes this runs... and you have not to close and open snap4arduino.
  - This issue depends on many details. How many devices do you have connected to this Arduino? Do you have servos? You comment on the external voltage... but is BT separate from servos? I think no. I think the jumper is to change between USB and an external source. Maybe your assembly needs to separate BT from servos power supply. Disconnect the servos and you'll see if this is the problem.

Joan

DenysZaytsev, 2017-05-27:
Hid jguille2,

I had similar issues and solved them using volt meter (based on resistor devider attached to one of the arduino Analog inputs and programmed with Snap to stop motors when voltage become lower then trashold). This helped, arduino is not rebooting and BT connection is okay. 

However I have another issue (just created a ticket for it). COM port disconnects randomly when script is executing (even when the script does nothing - empty cycle runs forever). If I leave it working over BT in a couple of minutes I'll get disconnected and will have to connect COM port again. I tried diferrent bluetooth modules - HC-05/HC-06 and SPP-CA/SPP-C based on diferrent chip, also tried different USB BT receivers on the laptop, but still nothing.

Can you help me with it?

closed 2023-05-25 by jguille2