bromagosa/Snap4Arduino #156: Snap4Arduino website status

Seven7au, 2017-05-04:
I am having trouble with accessing the Snap4Arduino website today. Wondering if the site is down or if it is on my end.
Also, one of my students and I have written a book using Snap4Arduino with the Arduino Uno and a shield that we designed. I have given Bromagosa a thank you in the book for help with the code. Please let me know where I can send you a copy.
Thanks, 
Seven

bromagosa, 2017-05-04:
I'm so sorry, there seems to be a problem with the server and they've told me they're doing some maintenance tasks. I hope it'll be back up soon!

Thank you so much for the book, please contact me via email at bernat at arduino dot org so I can send you my address :)

Seven7au, 2017-05-04:
Thanks for getting back so quickly Bromagosa.
I have made a quick Instructable here: [https://www.instructables.com/id/Banana-Keyboard-ARD2-INNOV8-and-Arduino/](url)
I will send you an email shortly.
Thanks

Seven7au, 2017-05-08:
Any further updates on the status of the website?

Is there an alternative link for the application downloads?

Thanks for your time with this.

Best regards, 
Seven

bromagosa, 2017-05-08:
I've set up a temporary mirror at https://bromagosa.github.io/Snap4Arduino/, but I haven't been able to push the releases yet. I'll be doing it this morning and let you know.

I'm so sorry about the situation, I'm doing everything I can but unfortunately the website is not under my control :(

bromagosa, 2017-05-08:
All the files have been uploaded now and. You can use https://bromagosa.github.io/Snap4Arduino/ as if it was the old site, or just get them from https://github.com/bromagosa/Snap4Arduino/releases/tag/1.2.3

Seven7au, 2017-05-08:
Thanks for the time that you have put into this. I hope that your service provider is able to fix the issue soon :)

closed 2017-05-11 by bromagosa