bromagosa/Snap4Arduino #158: 'prepare --all' fails with "no such remote ref e0ab485b7f8de567e7ecccc3a15ccabc619db446"

jarmopi, 2017-05-17:
Hello! I've been trying to build Snap4Arduino from sources, but after cloning the repo and commanding `./prepare --all` this is all I get:

```Pulling Snap! sources
error: no such remote ref e0ab485b7f8de567e7ecccc3a15ccabc619db446
Fetched in submodule path 'src/snap', but it did not contain e0ab485b7f8de567e7ecccc3a15ccabc619db446. Direct fetching of that commit failed.
ERROR: Could not set up Snap! git submodule.
Please make sure that you have access to the Internet and git is installed in your system.
```

I assure you I do have git installed and my internet connection is working perfectly :-) I'm using a 64-bit (amd64) Debian Stretch machine. node, npm, sed, git, wget, zip and unzip are all the latest versions.

`git submodule status` displays `+d811188e95ada17159f2461c347b8991cec6c7a6 src/snap (4.0.10.2-1-gd811188)`. `src/snap` is empty, and typing `git status` in it lists hundreds of deleted files that are uncommitted (?).

I can run `./prepare --nwjs` and `./prepare --desktop` fine, but without Snap, the build will ultimately fail.

Am I doing something wrong, or has something happened to Snap?

bromagosa, 2017-05-17:
Hi!

Snap! recently changed its git structure, so you'll need to fix your local copy before being able to pull it as a submodule for Snap4Arduino. Here's how to do it:

    git submodule deinit src/snap
    rm -r .git/modules/src

Now you should be able to pull it again with `./prepare --snap` or `./prepare --all` :)

Cheers,
Bernat.

jarmopi, 2017-05-17:
Hi! Unfortunately that does not work either.

The first command (git submodule) cleans the directory, but rm does not (the file does not exist). After this, `./prepare --all` still encounters the same problem.

This is a log of everything I've done, is it possible I'm missing some required step?

```
$ git clone https://github.com/bromagosa/Snap4Arduino.git
Cloning into 'Snap4Arduino'...
remote: Counting objects: 4762, done.
remote: Total 4762 (delta 0), reused 0 (delta 0), pack-reused 4762
Receiving objects: 100% (4762/4762), 45.41 MiB | 10.28 MiB/s, done.
Resolving deltas: 100% (1464/1464), done.
$ cd Snap4Arduino/
~/Snap4Arduino$ git submodule deinit src/snap
Cleared directory 'src/snap'
~/Snap4Arduino$ rm -r .git/modules/src
rm: cannot remove '.git/modules/src': No such file or directory
~/Snap4Arduino$ ./prepare --snap
Pulling Snap! sources
Submodule 'src/snap' (https://github.com/jmoenig/Snap--Build-Your-Own-Blocks) registered for path 'src/snap'
Cloning into '/home/XXXX/Snap4Arduino/src/snap'...
error: no such remote ref e0ab485b7f8de567e7ecccc3a15ccabc619db446
Fetched in submodule path 'src/snap', but it did not contain e0ab485b7f8de567e7ecccc3a15ccabc619db446. Direct fetching of that commit failed.
ERROR: Could not set up Snap! git submodule.
Please make sure that you have access to the Internet and git is installed in your system.
```

bromagosa, 2017-05-18:
Hi @jarmopi,

I'm travelling at the moment and can't really check it... maybe @jguille2 can help? I won't be able to test this out until Thursday next week :(

jguille2, 2017-05-19:
Hi @jarmopi and @bromagosa,
I think we must check our _prepare_ script and make some changes... but the main problem is the hash (commit) of the _snap submodule_. I don't find _e0ab485b7f8de567e7ecccc3a15ccabc619db446_ commit anywhere. Then we must change our base to the last snap release ( _eb8405c993e629f446c6336710492987d54f11bd_ ).

In the meantime, a workaround:
  - After a fresh cloning (or deleting 'src/snap' folder) we need to change the _submodule index_ to a 'valid' one
`
git update-index --cacheinfo 160000 eb8405c993e629f446c6336710492987d54f11bd src/snap
`
  - Then, you can run the _prepare_ script
`
./prepare --all
`
  - In the prepare process, there will be merging processes...(6). Then you will see the _merging dialog_. You only have to save the file and exit.

I think this is all.

Joan

jarmopi, 2017-05-19:
That works! I was able to build the program successfully. Thank you!

closed 2017-05-19 by jarmopi