bromagosa/Snap4Arduino #159: UDP server to interact with Reactivision

DavidGasku, 2017-05-17:
Hi, I was thinking on modify snap4arduino to include a UPD sever, in order to receive TUIO messages from Reactivision. This would make S4A able to interact with real life objects, moving sprites in the stage.

Do you think it would be feasible? Shoud I try to modify httpserver, or create a new UDP sever on its own.
Even if its not related to arduino, I think it will be easier to do this in S4A than Snap!, isn't it?

Thanks

DavidGasku, 2017-05-18:
Well, it's been easier than I thought. Using tuio-nw module, I was able to listen to Reactivision, and with some javascript blocks, move any sprite with any fiducial.
It's very rough, but I'll post my progress, in case anyone is interested.

bromagosa, 2017-05-18:
Cool! Yes please, do share it with us! :)

closed 2023-05-25 by jguille2